acados/acados #210: Panel major data in Input/Output structs of modules

nielsvd, 2018-02-15:
To avoid conversion between column major and panel major and backwards all the time, passing data between modules, it is preferred to define data in input/output-structs panel major by default.

Let's keep an overview of the pros/cons here and find remedies to problems that arise.

First move:
*  The parent module that the user created using the c-interface will have obfuscated input/output definitions. Do we want the user to know how to use blasfeo in order to set the fields, or do we define getters/setters for each and every input/output? In C++ this is less of an issue, since data conversion can be automated (hidden from user) when reading/writing to the members of the input/output-class.

dkouzoup, 2018-02-15:
@roversch already implemented setters and getters for the ocp_qp case, which are included in hpipm (since ocp_qp is part of hpipm). I suppose we should have the same getters and setters (as part of the c interface) for the other structs. As far as I am concerned this belongs to the interface and in the core we can assume that the user knows what he's doing.

dkouzoup, 2018-02-15:
An open question is what to do with sim_in, since erk uses doubles and irk structs.

giaf, 2018-03-29:
Probably not relevant any more. Otherwise, please open it again with updates

closed 2018-03-29 by giaf