acados/acados #211: Submodule structure

nielsvd, 2018-02-16:
## Sketch of prototype

``module_common.h``
```c
...

typedef struct module_config_ {
    int (*fun)(ocp_nlp_in *qp_in, ocp_nlp_out *qp_out, void *args, void *mem, void *work);
    int (*calculate_args_size)(ocp_nlp_dims *dims, void *submodules);
    void *(*assign_args)(ocp_nlp_dims *dims, void **submodules, void *raw_memory);
    void *(*copy_args)(ocp_nlp_dims *dims, void *raw_memory, void *source);
    void (*initialize_default_args)(void *args);
    int (*calculate_memory_size)(ocp_nlp_dims *dims, void *args);
    void *(*assign_memory)(ocp_nlp_dims *dims, void *args, void *raw_memory);
    int (*calculate_workspace_size)(ocp_nlp_dims *dims, void *args);
    void *submodules;
} module_config;

...
```

``module_impl1.h``
```c
...

typedef struct module_impl1_submodules_ {
    other_module_config *submodule1;
    another_module_config *submodule2;
} module_impl1_submodules;

typedef struct module_impl1_args_ {
    // some specific arguments
    ....

    // submodule config
    module_impl1_submodules submodules;

    // submodule arguments
    void *submodule1_args;
    void *submodule2_args;
} module_impl1_args;

...
```

## Explanation
*  The c-interface provides means to set up the recursive ``module_config``-struct. One of the supported approaches is the current ``module_plan`` configuration method (which should be renamed).
*  Using the ``module_config``-struct, one can obtain the ``module_impl1_args``-struct. The ``module_impl1_args``-struct holds a copy of the submodules struct. The submodules structs of the submodules will point to the copy of the submodules struct in the args struct of this specific submodule. E.g., ``module_impl1_args.submodules->submodule1->submodules = &module_impl1_args->submodule1_args.submodules``. This is implemented by overwriting the submodules pointer, when calling the ``assign_args()``-routine of the specific submodule, hence: notice the double-pointer to the submodules in ``void *(*assign_args)(ocp_nlp_dims *dims, void **submodules, void *raw_memory);``.
* The arguments now embed all data that is in the ``module_config``-struct. The original ``module_config``-struct can be freed.

dkouzoup, 2018-02-16:
My thoughts:

1. Didn't we decide to move the copy_args  to the interface instead?

2. I do not have a strong opinion on that, but the submodules struct could be an extra input to all functions, instead of copying them inside the arguments once created, which feels a bit of a hack. Would that make sense? To be honest I am a bit confused with the explanation of point 2 above.

3. I'd like to have the freedom to have implementation of submodules with a subset of the functions that you define above. Is that possible?

nielsvd, 2018-02-16:
1. Correct, it was moved to the interface. However, if the ``*_plan``-struct in the c interface is not the primary configuration strategy. But instead, a module is fully defined by the ``*_config``-struct, copying the arguments is the only missing feature in the core to make that work. Open for suggestions here..
2. I was thinking the same thing...
3. Yes, that is possible. However, note that submodules will only be compatible if their module definition is a subset of the functions that the parent module defines.

nielsvd, 2018-02-19:
Second iteration:

(note that I also added the dims-struct to the ``fun``-pointer to pass all data that was constructed to the function evaluation routine).

``module_common.h``
```c
...

typedef struct module_config_ {
    int (*fun)(module_dims *dims, module_in *in, module_out *out, void *submodules, void *args, void *mem, void *work);
    int (*calculate_args_size)(ocp_nlp_dims *dims, void *submodules);
    void *(*assign_args)(ocp_nlp_dims *dims, void *submodules, void *raw_memory);
    void *(*copy_args)(ocp_nlp_dims *dims, void *submodules, void *raw_memory, void *source);
    void (*initialize_default_args)(void *submodules, void *args);
    int (*calculate_memory_size)(ocp_nlp_dims *dims, void *submodules, void *args);
    void *(*assign_memory)(ocp_nlp_dims *dims, void *submodules, void *args, void *raw_memory);
    int (*calculate_workspace_size)(ocp_nlp_dims *dims, void *submodules, void *args);
    void *submodules;
} module_config;

...
```

``module_impl1.h``
```c
...

typedef struct module_impl1_submodules_ {
    other_module_config *submodule1;
    another_module_config *submodule2;
} module_impl1_submodules;

typedef struct module_impl1_args_ {
    // some specific arguments
    ....

    // submodule arguments
    void *submodule1_args;
    void *submodule2_args;
} module_impl1_args;

dkouzoup, 2018-02-19:
> Yes, that is possible. However, note that submodules will only be compatible if their module definition is a subset of the functions that the parent module defines.

Just keep in mind that this might make the number of function pointers of the parent blow up with time. For instance I can think of modules that could have several "funs", one corresponding to a different step of the module, within an algorithm.

roversch, 2018-02-20:
@dkouzoup Wouldn't these different function pointers belong to different modules, then?

dkouzoup, 2018-02-20:
No, that's the point. It could be different steps of the same module. Such that the init functions are common but the evaluation steps can be called separately.

dkouzoup, 2018-02-20:
Thinking about it a bit more, I would propose something more flexible/generic of the form:

```
typedef struct module_config_ {
    void *evaluation
    void *initialization
    void *submodules;
} module_config;
```

where "evaluation" struct has all module operations that take place online (e.g. "solve" for an ocp_qp), "initialization" struct has all helper functions (e.g. calculate_args_size/assign_args, or simply cost_calculate_size/cost_assign for the cost) and submodules follows the same reasoning as @nielsvd suggested. 

This would solve (at least I think) the issue of possibly multiple evaluate steps and would keep the creation of modules as minimal as required.

giaf, 2018-03-29:
Probably not relevant any more. Otherwise, please open it again with updates

closed 2018-03-29 by giaf