acados/acados #212: Distinction sim_solver_config, sim_input_config.

nielsvd, 2018-02-16:
Currently in the following lines of code, the ``sim_solver_config`` struct defines both the function pointers necessary to create the solver-specific data and also to create the input struct:
https://github.com/acados/acados/blob/1d5dcdfac7744f9c22b367c0ff77edbc22ab98de/acados/sim/sim_common.h#L125

Given that it is desired to make a distinction between the NLP definition and the solver definition, it would make sense to split the ``sim_solver_config`` struct and to create for intance a ``sim_input_config``-struct.

dkouzoup, 2018-02-16:
Alternatively, can we not use the sim_solver_config to have all submodules and use some to build the input struct and some the solver? This would also avoid the possibility to build an inconsistent input for a sim_solver (since we will be passing the same submodules to both). If it is better from the interface perspective, I do not mind the separation either.

giaf, 2018-03-29:
Probably not relevant any more. Otherwise, please open it again with updates.

closed 2018-03-29 by giaf