acados/acados #217: assert being overwritten by qpoases/qpdunes

dkouzoup, 2018-02-23:
This is particularly "dangerous" since in many places, asserts are not actually checked because external software has overwritten it with an empty function..

doanminhdang, 2018-02-23:
Isn't this has similar potential impact like removing the flag -Werror in compilation?
Will there be a lot of bugs if the overwritten empty functions are removed?

dkouzoup, 2018-02-26:
> Isn't this has similar potential impact like removing the flag -Werror in compilation?
> Will there be a lot of bugs if the overwritten empty functions are removed?

not sure I understand what you mean

dkouzoup, 2018-02-26:
closing for now, I was not compiling with Debug on.. But there is still a chance our assert is overwritten by qpoases in some places (qpoases interface?).

closed 2018-02-26 by dkouzoup