acados/acados #260: introduce "memsize" in all structs to make cast cheaper

giaf, 2018-04-11:
As @roversch noted some time ago, cast_workspace is expensive, and it is happening on-line, every time a routine is called.

A way to make it cheaper is to add a member "memsize" to all structs, to avoid to call calculate_size on-line for all sub-modules.

giaf, 2018-04-11:
@dkouzoup maybe a task for Branimir?

dkouzoup, 2018-04-11:
duly noted!

dkouzoup, 2018-04-18:
@giaf: what about assign_and_advance instead? Are there any disadvantages you can think of?

giaf, 2022-11-30:
Getting back to this a few years later, as asked by @FreyJo 

Looking e.g. at the code https://github.com/acados/acados/blob/master/acados/ocp_nlp/ocp_nlp_sqp.c#L412 it can be seen that for each sub-module the amount of workspace is compute three times: 1) inside the `assign` function (e.g. https://github.com/acados/acados/blob/master/acados/ocp_nlp/ocp_nlp_sqp.c#L423) ; 2) inside the `calculate_size` function (e.g. https://github.com/acados/acados/blob/master/acados/ocp_nlp/ocp_nlp_sqp.c#L424) ; 3) the workspace of the entire module is computed once more for the final assert (e.g. https://github.com/acados/acados/blob/master/acados/ocp_nlp/ocp_nlp_sqp.c#L445 ), and this happens every time the main module is called (i.e. at least at each OCP problem solution, if not more often).

As suggested by @dkouzoup One way to remove one computation is to have an `assign_and_advance`, where the pointer to the raw memory is advanced internally without the need to call the `calculate_size` function of the sub-module once more.
One more computation can be clearly removed by avoiding the final assert.
This is easy and quick to do without affecting the functioning of the acados core.

However, since we do not allow the problem size to change anyway at different OCP problems solutions, it could be possible to compute these only at the very first OCP problem solution and to keep the needed quantities cached in the memory.
E.g. the memory struct could cache the workspace struct (e.g. https://github.com/acados/acados/blob/master/acados/ocp_nlp/ocp_nlp_common.c#L1767) such that at the end of the day only the actual workspace of numerical values (e.g. arrays of integers or floats) end up in the workspace raw memory, while all structs around them are kept in memory.
However this would require a quite involved change of the acados core.

sandmaennchen, 2024-05-06:
Mostly addressed here: https://github.com/acados/acados/pull/888

closed 2024-05-06 by sandmaennchen