acados/acados #268: keep updating the file convention_about_external_functions.txt

giaf, 2018-04-19:
Everyone adding/editing an external function call in acados should _immediately_ update the file
'''convention_about_external_functions.txt'''
accordingly. (at least until this finds its way into a better documentation)

@zanellia can you add the new_lifted_irk? (by the way, are you going to do for them the same that Jonathan did for irk?)
@roversch can you add the quadratic-over-nonlinear constraints, once fixed?

Thanks everyone!

zanellia, 2018-04-20:
>
> @zanellia <https://github.com/zanellia> can you add the new_lifted_irk?
> (by the way, are you going to do for them the same that Jonathan did for
> irk?)
>

Yes, and yes :)

giaf, 2018-04-21:
great :) thanks @zanellia !

FreyJo, 2020-10-07:
The file was moved here:
https://github.com/acados/acados/blob/4dd0f775450af9d47615f985066c923c3d2b51c2/utils/convention_about_external_functions.txt

NOTE: It does not seem to be up to date regarding e.g. missing parameters, BGP, external functions for cost evaluation.
Anyone who is actually interested in the external functions convention should look at the generating files in the Matlab and Python interfaces which will stay up to date, since they are tested on Travis.

closed 2020-10-07 by FreyJo