acados/acados #269: wind_turbine_nmpc_example broken?!

FreyJo, 2018-04-20:
Hey guys,

for me the wind_turbine_nmpc_example fails during the ctest. I looked more into it and also ran it manually:

I went back some commits, always running "submodule update" before compiling with cmake, and found out the following:

In the commit c1cdf67f5913fcacd3dde3dde7d0e505173fc8a8, April 12th,  the example works fine for me running all 40 problems and putting out

`[...] 
success!`

In the commit d8a6a2cc9f504fe14d31504dd3e9ba2eb5058c6d, April 12th, the example stops working for me.
It outputs:
```
problem #0, status 1, iters 10
xsim = 
  1.29820   0.00000  50.19698   0.00000   6.00279   4.33119   6.00279   4.33119 

electrical power = 5.148647

residuals
res_g =
3.925749e-12	6.518341e-12	5.456968e-11	6.184564e-11	-1.172396e-13	1.127773e-10	5.883294e-12	-9.441337e-13	1.287361e-10	1.391243e-11	[...]
```

Interestingly, in more recent versions (c9716900d378c6a41bdd861ef1e4dc632765478c - from April 19th), the example fails too, but with another output:

```
problem #0, status 0, iters 9
xsim = 
  1.29820  -0.00000  50.19698   0.00000   6.00279   4.33119   6.00279   4.33119 

electrical power = 5.148647

problem #1, status 1, iters 10
xsim = 
  1.28755   0.02942  50.45517   0.00342   6.34369   4.51163   7.60279   4.53119 

electrical power = 5.319151

residuals
res_g =
4.096812e-11	3.568517e-12  [...]
```
It seems like here the first problem runs fine but in the second one the solver fails and returns status 1. So it seems that there has been a commit between 12th and 19th of April which also changed the behaviour of the ocp_nlp_solve..

@giaf  @dkouzoup : Do you have ideas where the problem is originated?

FreyJo, 2018-04-20:
UPDATE:
I found out the example does not crash when changing

`		plan->sim_solver_plan[i].sim_solver = NEW_LIFTED_IRK;`

to

`		plan->sim_solver_plan[i].sim_solver = IRK;`

@zanellia Do you have an idea, why using this integrator causes problems?

zanellia, 2018-04-20:
@FreyJo, it seems to work fine with MAX_SQP_ITERS >= 14 when using
plan->sim_solver_plan[i].sim_solver = NEW_LIFTED_IRK;

giaf, 2018-04-21:
The combination of lifted_irk and soft-constraints makes the problem a bit unstable numerically. Very small differences in the numerics (like e.g. the different BLASFEO and HPIPM versions used by default in make and cmake) affect the number of SQP iterations needed to fully converge.

The easy solve it to increase the max number of SQP iterations e.g. to 15.

dkouzoup, 2018-04-23:
output on mac (10 iterations are enough):

problem #0, status 0, iters 9
xsim =
  1.29820  -0.00000  50.19698   0.00000   6.00279   4.33119   6.00279   4.33119

electrical power = 5.148647

problem #1, status 0, iters 9
xsim =
  1.28755   0.02942  50.45517   0.00342   6.34369   4.51163   7.60279   4.53119

electrical power = 5.319151

problem #2, status 0, iters 9
xsim =
  1.30247   0.02449  50.71378   0.00932   7.18001   4.56581   9.20279   4.57004

electrical power = 5.445398

problem #3, status 0, iters 9
xsim =
  1.33201   0.00016  50.97716   0.01192   8.31681   4.46407  10.80279   4.45268

electrical power = 5.444783

problem #4, status 0, iters 6
xsim =
  1.35615  -0.01739  51.24619   0.01002   9.63587   4.37825  12.40279   4.36991

electrical power = 5.436894

problem #5, status 0, iters 4
xsim =
  1.36136  -0.02738  51.51833   0.00516  10.84025   4.18965  12.94574   4.16991

electrical power = 5.222682

problem #6, status 0, iters 4
xsim =
  1.34604  -0.01580  51.78933   0.00057  11.49850   4.05667  12.14694   4.04392

electrical power = 5.000024

problem #7, status 0, iters 4
xsim =
  1.32270   0.00174  52.05621  -0.00084  11.60018   4.03688  11.42667   4.03581

electrical power = 4.889355

problem #8, status 0, iters 5
xsim =
  1.30629   0.01648  52.31889   0.00122  11.39590   4.04730  10.78830   4.04852

electrical power = 4.841163

problem #9, status 0, iters 4
xsim =
  1.30388   0.01277  52.57970   0.00436  11.04160   4.07445  10.24751   4.07729

electrical power = 4.864636

problem #10, status 0, iters 5
xsim =
  1.30863   0.00025  52.84093   0.00566  10.63391   4.11053   9.80049   4.11421

electrical power = 4.925611

problem #11, status 0, iters 4
xsim =
[... 150 lines cut ...]
  1.26375  -0.00287  59.49100   0.00441   8.36600   4.24871   8.37997   4.24257

electrical power = 4.916554

problem #37, status 0, iters 4
xsim =
  1.26154   0.00081  59.74352   0.00421   8.33646   4.14661   8.21551   4.13605

electrical power = 4.790007

problem #38, status 0, iters 4
xsim =
  1.26263   0.00638  59.99583   0.00503   8.23427   4.01417   7.95928   4.00070

electrical power = 4.641043

problem #39, status 0, iters 4
xsim =
  1.26957   0.00413  60.24897   0.00615   8.05781   3.87165   7.63887   3.85734

electrical power = 4.500875

total time (including printing) = 647.364187 ms (time per SQP = 16.184105)

success!

zanellia, 2018-04-23:
So, recap. Convergence can be achieved in < 10 iterations when using the GENERIC target on Mac, but not on Linux. Convergence can be achieved in < 10 iterations when using the HASWELL target on Linux.

closed 2024-05-10 by FreyJo