acados/acados #271: MATLAB segfaults when QP solver is not compiled

dkouzoup, 2018-04-23:
As an example, on windows we do not compile HPMPC (due to variable size arrays). Matlab segfaults with the following info instead of simply throwing an error that solver does not exist:

```
Stack Trace (from fault):
[  0] 0x0000000000000000                                   <unknown-module>+00000000
[  1] 0x00000000e30fce92 ..\lib\acadosMEX.mexw64+00183954 ocp_qp_partial_condensing_solver_opts_calculate_size+00000034
[  2] 0x00000000e30edcb0 ..\lib\acadosMEX.mexw64+00122032 ocp_qp_opts_create+00000016
[  3] 0x00000000e30f773d ..\lib\acadosMEX.mexw64+00161597 ZN6acados6ocp_qp17initialize_solverESsSt3mapISsPNS_8option_tESt4lessISsESaISt4pairIKSsS3_EEE+00000189
[  4] 0x00000000e30ea7ca ..\lib\acadosMEX.mexw64+00108490 Z38_wrap_ocp_qp_initialize_solver__SWIG_0iPP11mxArray_tagiS1_+00001274
[  5] 0x00000000e30ea9f0 ..\lib\acadosMEX.mexw64+00109040 Z30_wrap_ocp_qp_initialize_solveriPP11mxArray_tagiS1_+00000144
[  6] 0x00000000e30ec5ae ..\lib\acadosMEX.mexw64+00116142 mexFunction+00000398
[  7] 0x00000000fc60234a                               bin\win64\libmex.dll+00140106 mexRunMexFile+00000314
 ```

roversch, 2018-04-23:
@dkouzoup I pushed a fix in https://github.com/acados/acados/issues/271 . 

Could you check if it works?

dkouzoup, 2018-04-24:
Great!

closed 2018-04-24 by dkouzoup