acados/acados #277: swig error with matlab

bbrito, 2018-04-25:
I'm trying to compile acados with Matlab2017b. It the Matlab flag is set to zero everything compiles fine.

But when I try to compile the Matlab modules I got the following error:

swig error : Unrecognized option -matlab
Use 'swig -help' for available options.
swig/matlab/CMakeFiles/acadosMATLAB.dir/build.make:61: recipe for target 'swig/acadosMATLAB_wrap.cxx' failed
make[2]: *** [swig/acadosMATLAB_wrap.cxx] Error 1

Any idea how to solve this? I check and cmake can find Matlab without a problem.

dkouzoup, 2018-04-25:
Hi,

The problem is that cmake found your system's swig and _not_ the one shipped with acados (in the folder external, which you need to compile first, following the installation instructions).

_However_, keep in mind that the matlab interface is currently not working for integrators and NLPs, only for QPs (we are working on fixing this issue, the interface is broken after some major changes in the software).

Cheers

bbrito, 2018-04-25:
Ahhh I see... I will try it... 

Thanks for the quick reply!

Cheers

closed 2018-11-20 by FreyJo