acados/acados #284: Running the ocp_nlp in the README

molysgaard, 2018-05-02:
Hi, I tried following the instructions in the README on how to get acados up and running.

I got all the submodules and external libraries correctly, and I got swig right.
Everything compiled correctly as well.
I got the PYTHONPATH correctly and i checked that the `make install` created an `acados.py` file.
I am able to run the example:
```
python examples/python/ocp_qp.py
```
but when I run the example in the readme:
```
python examples/python/ocp_nlp.py
```
I get the following error:
```
ImportError: cannot import name 'ocp_nlp_function'
```
On investigation I found that the `acados.py` file generated by swig does not contain any `ocp_nlp_function`. Is the readme outdated, or is there something I should do to generate the correct interface for the ocp_nlp?

dkouzoup, 2018-05-02:
Hi,

We are sorry for the outdated information in the readme. The NLP swig interface is currently undergoing some major changes and it is therefore broken. Indeed only the QP interface is working for the moment. We are working on it.

Thanks for the understanding

dkouzoup, 2018-05-02:
PS: All functionality is there using the c interface

dkouzoup, 2018-05-02:
Indeed, the way we do it at the moment is using casadi from matlab to code generate all the functions required by e.g. the integrators and then manually link them in C. 

See, e.g., the nonlinear_chain_ocp_nlp.c and the corresponding code-generation script under implicit_chain_model/chain_mass_impl.m 

Of course once the matlab interface is back, one can simply define the model in casadi symbolics and the generation of the appropriate function will take place in the background. Stay tuned ;)