acados/acados #317: Pointer alignment on 32-bit systems

roversch, 2018-08-09:
Throughout the code, we have checks that pointers should be aligned to 8 bytes. Obviously, this doesn't work for 32-bit systems, as an uneven amount of pointers to be 'assigned' will result in assertion errors.

We should make the alignment requirement match the type of system we are running on.

dkouzoup, 2018-09-04:
I am trying to work a bit on that. Question: is the size of the void pointer always at least as big as the size of any other pointer? I would assume (and partially read) so, since if sizeof(void *) = 4 and sizeof(double *) = 8 it would not be possible to store a double pointer value to a void pointer. Can someone confirm this? @giaf @roversch @tmmsartor

roversch, 2018-09-04:
In the C standard, it says (N1570 page 43, underlining mine):

"

   1.
      1.

      *A pointer to void **shall have the same representation and alignment
      requirements as a pointer to a character type*.48) Similarly,
      pointers to qualified or unqualified versions of compatible types shall
      have the same representation and alignment requirements. All pointers to
      structure types shall have the same representation and alignment
      requirements as each other. All pointers to union types shall
have the same
      representation and alignment requirements as each other. *Pointers to
      other types need not have the same representation or alignment
      requirements.*

      ...

      48) The same representation and alignment requirements are meant to
      imply interchangeability as arguments to functions, return values from
      functions, and members of unions.

"

From this, I understand that pointers to double could be smaller, equal or
larger than pointers to void.

http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1570.pdf

dkouzoup, 2018-09-04:
Ok thanks for the quote, but in that case I don't understand how we can store a value of a double pointer to a void pointer safely. I mean in acados we have void pointers all over the place inside structures and when we calculate the sizeof(struct), the size of void pointer is taken into account. If later on we store in this field the value of a double pointer how can this work?

roversch, 2018-09-04:
Same document, page 55:

"

   1.

   A pointer to void may be converted to or from a pointer to any object
   type. A pointer to any object type may be converted to a pointer to void and
   back again; the result shall compare equal to the original pointer.

"

So you can use void pointers; just don't make assumptions on their size.

roversch, 2018-09-04:
By the way, I found this SO post helpful:

https://stackoverflow.com/questions/20338147/void-pointers-pretending-to-be-void-double-pointers

roversch, 2018-09-05:
> Yes indeed. I am still confused though as on how it is possible not to
> loose information when converting a double pointer to void and back to
> double if the double pointer needs double the bytes. Does this mean that
> these extra bytes are redundant and information is not lost during the
> conversion?

How it is exactly done is depending on what platform and what compiler
you're using. Remember that you cannot dereference a void pointer, so their
representation doesn't have to map to physical memory addresses (nor do any
other pointers have to by the way). The compiler can insert all kinds of
addressing arithmetic for the conversion from pointer-to-double to
pointer-to-void and back.

dkouzoup, 2018-09-06:
Thanks! That was quite helpful