acados/acados #330: Undefined behavior due to alignment

roversch, 2018-09-04:
When running e.g. the `nonlinear_chain_ocp_nlp_example` with `-fsanitize=undefined` compiler flags for Clang, I get many errors concerning undefined behavior, mainly of the type

- member access within misaligned address
- load of misaligned address

For example, in line 

https://github.com/acados/acados/blob/70a9a83fb4fdacd0d7429fe6acc75bce40589c64/acados/ocp_nlp/ocp_nlp_dynamics_cont.c#L64 

`sim` is not aligned to 8 bytes, but to 4. This is what invokes the undefined behavior. The error messages are quite helpful, these errors could be resolved by putting extra alignment points in `calculate_size` and `assign` functions.

dkouzoup, 2018-09-05:
Nice! This is very helpful, thanks for reporting

dkouzoup, 2018-09-05:
@roversch, is there anything special I need to do to enable this? By adding it in the CMAKE_C_FLAGS I get many errors like this one:

Undefined symbols for architecture x86_64:
  "___ubsan_handle_add_overflow", referenced from:
      _main in dense_qp.c.o
      _blasfeo_allocate_dmat in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      _blasfeo_allocate_dvec in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      _blasfeo_print_dmat in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      _blasfeo_print_to_file_dmat in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      _blasfeo_print_to_file_exp_dmat in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      _blasfeo_print_to_string_dmat in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      ...
  "___ubsan_handle_divrem_overflow", referenced from:
      _print_ocp_qp_info in libacados.a(print.c.o)
      _make_int_multiple_of in libacados.a(mem.c.o)
      _align_char_to in libacados.a(mem.c.o)
      _QProblemB_hotstart in libqpOASES_e.a(QProblemB.c.o)
      _QProblemB_updateFarBounds in libqpOASES_e.a(QProblemB.c.o)
      _QProblemB_backsolveRrem in libqpOASES_e.a(QProblemB.c.o)
      _QProblemB_performRatioTestB in libqpOASES_e.a(QProblemB.c.o)
      ...
  "___ubsan_handle_float_cast_overflow", referenced from:
      _DenseMatrix_times in libqpOASES_e.a(Matrices.c.o)
      _DenseMatrix_transTimes in libqpOASES_e.a(Matrices.c.o)
      _spotrf_ in libqpOASES_e.a(Matrices.c.o)
Undefined symbols for architecture x86_64:
  "___ubsan_handle_add_overflow", referenced from:
      _blasfeo_allocate_dmat in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      _blasfeo_allocate_dvec in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      _blasfeo_print_dmat in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      _blasfeo_print_to_file_dmat in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      _blasfeo_print_to_file_exp_dmat in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      _blasfeo_print_to_string_dmat in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      _blasfeo_print_exp_dmat in libblasfeo.a(d_aux_ext_dep_lib4.c.o)
      ...

dkouzoup, 2018-09-05:
It seems it's a C++ flag? But when I compile the examples with C++ I get other errors about malloc etc

roversch, 2018-09-05:
I just edited these two lines:

```
    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=c99 -Wall -fdiagnostics-show-option -fsanitize=undefined")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -Wall -fdiagnostics-show-option -fsanitize=undefined")
```

dkouzoup, 2018-09-05:
Lol, I tried to add it in either only CMAKE_C_FLAGS, or CMAKE_CXX_FLAGS and that didn't work :D 

Now I am stack to 100% build for the last 10 min, was it the same for you?

dkouzoup, 2018-09-05:
Done!

dkouzoup, 2018-09-05:
I am on it.

roversch, 2018-09-05:
Yeah, the builds take a long time, especially for the long CasADi-generated files. UBSan has to add instructions at every call that could potentially invoke undefined behavior.

dkouzoup, 2018-09-06:
Fixed everything on our current examples in my open PR. They were not many errors, just the same things being propagated. However, I guess other examples would trigger other undefined behaviors, so at some point we should clean up and be consistent in all calc_size assign routines..

dkouzoup, 2018-09-06:
So I would leave this issue open for the time being

FreyJo, 2024-05-10:
I am quite certain that this is resolved.

closed 2024-05-10 by FreyJo