acados/acados #407: Setting up LD_LIBRARY_PATH for Matlab interface

jkoendev, 2019-07-17:
I think it is undesirable to have to write and run/source an `env.sh` for each problem that you implement in Matlab. This disturbs the workflow a lot (also if used from OpenOCL) because you have to restart Matlab, and run from the command line after you implemented your model.

I investigated and tried writing a Matlab `.m` script that sets the environment variables in the same way but unfortunately setting LD_LIBRARY_PATH does not work as intended. I think Matlab only uses LD_LIBRARY_PATH at the startup to look for runtime libraries (.so), and it does not have an effect if you change the path later. I have not found any workaround for that.

An option would be to use the matlab `system` command to set the environment variables, and restart matlab but this might have to many side effects (not sure if it works for all configurations). The command would be something like:
```
system('export LD_LIBRARY_PATH=...')
system([fullfile(matlabroot,'bin','matlab'), ' &']);exit
```

Another option would be to setup a runtime directory during compilation of acados, e.g. `acados/runtime` or using `acados/lib`, add it to LD_LIBRARY_PATH forever and copy all the .so/.mex files there. So the `libacados.so`, `libblasfeo.so` etc. but also the generated model files like `libocp_model.so`, `ocp_create.mexa64` etc. These need to have a unique name for each model then (which is probably a good idea anyway because there could be conflicts if multiple models are added to the path). This would be my preferred way atm.

jkoendev, 2019-07-17:
setting ld path more permanently on Linux (outside of bash) might involve `ldconfig`.

Another option would be to use linker arguments `-Wl,-rpath,/path/to/runtime/dir`.

giaf, 2019-07-18:
some inspiration maybe
https://tomopt.com/docs/instguide/tomlab_install006.php

closed 2019-08-07 by jkoendev