acados/acados #411: Building acados on windows with mex interface

jkoendev, 2019-07-23:
Ill post my progress on building acados on Windows 10 here.

I was able to compile acados with Cmake and Visual Studio 2019 but as Matlab does not support new msvc compilers i will try MinGW.

zanellia, 2019-07-23:
You don't actually need MATLAB to support the compiler that you use to generate the shared library right? Or am I missing something?

jkoendev, 2019-07-23:
step inspired from [here](https://github.com/tudelft3d/masbcpp/wiki/Building-on-Windows-with-CMake-and-MinGW)
1. download and execute the [CMake Win64 Installer](http://www.cmake.org/download/). Make sure to set the PATH variable during installation
1. download and install [mingw-w64](http://mingw-w64.yaxm.org/doku.php/download/mingw-builds). Select x86_64 instead of i686 in the options
1. add the mingw-w64 programs to the system PATH variable (eg. append this string `D:\software\mingw-w64\x86_64-8.1.0-posix-seh-rt_v6-rev0\mingw64\bin`)

jkoendev, 2019-07-23:
I am getting linker error because Matlab MinGW compiler can not link to libacados.lib which was generated from msvc

jkoendev, 2019-07-23:
Open Powershell and run 
```
mkdir build
cd build
cmake .. -G "MinGW Makefiles"
mingw32-make
```

jkoendev, 2019-07-23:
adapted paths in `ocp_compile_mex.m`: 66d0f7f578a3b5e5bdad7fe71c9483c83b759320

jkoendev, 2019-07-23:
651d92644709fbf30f7fa189156fb511b96314fb

jkoendev, 2019-07-23:
3d251ebc8f2d63fcd3fd02c31fa4cc4ec32fc797

jkoendev, 2019-07-23:
most things are compiling fine now!
still need to make [this file](https://github.com/acados/acados/blob/master/interfaces/acados_matlab/sim_set_ext_fun.m) windows compatible.

closed 2019-08-01 by jkoendev

jkoendev, 2019-09-13:
on windows build static library:
```
cmake.exe -G "MinGW Makefiles" -D BLASFEO_TARGET=GENERIC -D HPIPM_TARGET=GENERIC -D ACADOS_INSTALL_DIR=../acados_install -DBUILD_SHARED_LIBS=OFF ..
```

until #471 which sets static library as default on windows

RoyiAvital, 2020-03-19:
The issue is static libraries built by MinGW can't be used (Or in high percentage) with MSVC / Intel Compiler Projects.

Maybe DLL's works better, I haven't checked.

FreyJo, 2020-03-20:
@RoyiAvital 
the latest workflow to use `acados` Matlab on Windows has been documented here:
https://docs.acados.org/installation/index.html#windows-for-use-with-matlab