acados/acados #412: Mex examples on Octave 4.2.2

jkoendev, 2019-07-24:
I needed to replace the `import casadi.*` because it is not supported on Octave 4.2.2

Then i get this error when running `linear_mass_spring_model/example_ocp.m`

```
compiling /home/jonas/repos/acados_octave/examples/matlab_mex/linear_mass_spring_model/../../../interfaces/acados_matlab/ocp_create.c
warning: mkoctfile: building exited with failure status
In file included from /home/jonas/repos/acados_octave/examples/matlab_mex/linear_mass_spring_model/../../../acados/ocp_qp/ocp_qp_common.h:25:0,
                 from /home/jonas/repos/acados_octave/examples/matlab_mex/linear_mass_spring_model/../../../acados/ocp_nlp/ocp_nlp_constraints_common.h:31,
                 from /home/jonas/repos/acados_octave/examples/matlab_mex/linear_mass_spring_model/../../../acados/ocp_nlp/ocp_nlp_common.h:42,
                 from /home/jonas/repos/acados_octave/examples/matlab_mex/linear_mass_spring_model/../../../interfaces/acados_c/ocp_nlp_interface.h:28,
                 from /home/jonas/repos/acados_octave/examples/matlab_mex/linear_mass_spring_model/../../../interfaces/acados_matlab/ocp_destroy.c:6:
/home/jonas/repos/acados_octave/examples/matlab_mex/linear_mass_spring_model/../../../external/hpipm/include/hpipm_d_ocp_qp.h:35:10: fatal error: blasfeo_target.h: No such file or directory
 #include <blasfeo_target.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

zanellia, 2019-07-24:
I guess you are missing casadi-octave? `import` should be implemented there.

giaf, 2019-07-24:
yes if `casadi-octave` is in the matlab path, the implementation of `include` in there is used

giaf, 2019-07-24:
I also use octave 4.2.2

jkoendev, 2019-07-24:
ok right the `import` works now, not sure what was wrong before, prob. messed up the path with a second casadi installation. And the blasfeo error also goes on me, made a fresh clone of acados and forgot to compile ;)

closed 2019-07-24 by jkoendev