acados/acados #414: Mex interface, time optimal control

jkoendev, 2019-07-24:
Is it enough to just not set `ocp_model.set('T', T)`? 
I think getting the solution for the final time using `ocp.get('T')` is not yet implemented.

jkoendev, 2019-07-24:
i tried getting:
```
error: ocp_create: T not set!
```

zanellia, 2019-07-24:
T is not an optimization variable in acados at the moment. Time-optimal formulations can be implemented using time transformations instead.

jkoendev, 2019-07-24:
ok thanks! The transformation does not seem trivial to me for a general problem. Have you/somebody got some code to do the coordinate transformation for some problem? Do you have a model for a time-transformed cartpole-swingup somewhere (time optimal swingup)?

closed 2019-08-01 by jkoendev