acados/acados #415: Mex interface, setting empty constraints lead to crash

jkoendev, 2019-07-24:
One of these
```
  ocp_model.set('constr_expr_h_e', []);
  ocp_model.set('constr_lh_e', []);
  ocp_model.set('constr_uh_e', []);
```
leads to a mex creash
```
--------------------------------------------------------------------------------
       Integer divide by zero detected at Wed Jul 24 11:11:40 2019 +0200
-------------------------------------------------------------------------
Caught MathWorks::System::FatalException
```

jkoendev, 2019-07-24:
[matlab_crash_dump.3268-2.txt](https://github.com/acados/acados/files/3428800/matlab_crash_dump.3268-2.txt)

zanellia, 2019-07-24:
This will be fixed when we add code-gen to the MATLAB interface. Otherwise, separate checks need to be implemented.

closed 2019-08-01 by jkoendev