acados/acados #425: Binaries for Windows and Linux

jkoendev, 2019-08-05:
I am now maintaining binaries of acados for Windows and Linux (64 bit) in this repo:
https://github.com/jkoendev/acados-deployment/releases

OpenOCL will download the binaries from there automatically to set up acados for the user.

The archives also contain the source code of acados (and external packages) with the license text so i think the license requirements should be satisfied.

If you want you can link to or redistribute the archives in the `acados` releases.

zanellia, 2019-08-05:
That's great! I guess we could do this directly using GitHub releases, or is it already doing so? I would be interested in using the same framework for the binaries of the template parser :)

jkoendev, 2019-08-05:
I am using github releases but it is on my personal repo. You can download the .zip files from there and upload them to the official acados releases.

jkoendev, 2019-08-05:
btw as acados does not have a version yet, i used the git commit id as the version identifier. Would be nicer if acados had an official version like v1.0 or so..

closed 2019-08-07 by jkoendev

BoostCookie, 2026-10-12:
Binary releases on the original git repo would be nice. Maybe a CI job could do it for each commit that has a version tag?