acados/acados #427: ACADO vs ACADOS

ctoumieh, 2019-08-06:
Hi,
I have been using ACADO for some time to control a UAV (quadrotor). With ACADOS, I used the same model and same parameters (to an extent) but the control isn't working. I don't have a problem in the model (verified it in another way) and I don't think I have a formulation error (you can see for yourself in the zip). How different is ACADO from ACADOS (I know ACADOS is built on casadi but is it as fast/robust
[uav_code.zip](https://github.com/acados/acados/files/3474521/uav_code.zip)
)?
Cheers,
CT

FreyJo, 2019-08-07:
Hi,
first: [Difference between acados and ACADO](https://discourse.acados.org/t/what-is-the-difference-between-acados-and-acado/19)
In terms of speed acados is very similar to ACADO. We submitted a conference paper on acados with a benchmark, which is however not public yet as far as I know.

I will have a look into your Code soon and answer agian.

zanellia, 2019-08-07:
Hi CT,

I was able to run your script, but I cannot interpret the results of the
closed-loop simulation. The QP solver fails at some point, which is
definitely not a good sign :) However, using SQP_RTI as an NLP solver, the
simulation goes through without errors.

Maybe a few more details about how you were using ACADO could help to
justify the discrepancy in the results:

1) Were you using ACADO code-gen or not? (the SQP solvers are implemented
differently)
2) Were you suing RTI or solving to local optimality?

In general, I would recommend to use the acados forum for this kind of
discussions: https://discourse.acados.org/.

Cheers,
Andrea

ctoumieh, 2019-08-07:
Hi,
If you use SQP_RTI and look at the plot, the inequality constraints for the command are violated. I was using ACADO with code generation and RTI I think. From now on I will move to the forum if I have further questions.
Cheers,
CT

ctoumieh, 2019-08-14:
It was a typo in idxbu. My bad. Thanks for your help.

closed 2019-08-14 by ctoumieh