acados/acados #428: Mex setting maximum iterations

jkoendev, 2019-08-07:
Can i set the maximum iterations after the solver object was created somehow? 

e.g. 
```
ocp = acados_ocp(ocp_model, ocp_opts);
ocp.set('nlp_solver_max_iter', 1)
```

@giaf

FreyJo, 2019-08-07:
Have you tried this?
`ocp_opts.set('nlp_solver_max_iter', nlp_solver_max_iter);`
As far as I understand, this setter will set the option in the C object, which is used by the ocp_solver internally, created by
`ocp = acados_ocp(ocp_model, ocp_opts);`

jkoendev, 2019-08-07:
i just tried and it didn't work! In the pendulum on a cart example i did
```
ocp_opts.set('nlp_solver_max_iter', 4);
ocp.solve();
```
but i t still does 9 iterations

giaf, 2019-08-07:
That won't work, because it just changes the value locally in the `ocp_opts`.

The way to go is to implement `ocp.set('nlp_solver_max_iter', 1)`

FreyJo, 2019-08-07:
will be merged soon https://github.com/acados/acados/pull/431

closed 2019-08-07 by FreyJo

giaf, 2019-08-07:
here it is https://github.com/acados/acados/commit/44c215e246b1dc92ace600c37a05d97265ae658d

giaf, 2019-08-07:
@FreyJo ops I also did it now... sorry I didn't see you PR...
BTW the matter was slightly more complicated, see commit above