acados/acados #482: [Python interface] acados_solver.set(stage,field,value) and get() do not work properly

fbailly, 2019-10-01:
Hi everybody,

Thank you for the work done here !

I am generating an OCP in ACADOS using the python interface. I want to warm-start the solver with an initial guess on the state "x" and on the control "u". I looks like the get() and set() methods of the solver instance are the right tools to do so. However, they don't seem to work properly :

```
In [6]: acados_solver.set(0,"u",np.array([1,1]))                                                        
In [7]: acados_solver.get(0,"u")                                                                        
Out[7]: array([0., 0.])
```
Which of the setter or the getter is not working, i.e. even if the output is not the one expected, is the set() taken into account ?

Thank you !

zanellia, 2019-10-01:
Hi @fbailly,

First of all thanks for your interest in acados. There seems to be an issue with the setter you are using: setter for `x` and `u` were not actually implemented, but no error was thrown due to a silent defaulting behavior. I have fixed this and implemented setters for `x` and `u` so you can provide an initialization (see https://github.com/acados/acados/pull/483). Let me know if this solves your problem ;)

fbailly, 2019-10-01:
@zanellia , thanks this patch does work !

zanellia, 2019-10-01:
Great, closing this :).

closed 2019-10-01 by zanellia