acados/acados #517: [Python] nothing works if put outside acados source directory

Maik93, 2019-11-24:
Hi,
just installed acados, I'm trying to make it works with Python and CasADi.

Currently I've been able to run examples contained in `examples/acados_python/` by adding this lines in my bashrc:
```bash
export ACADOS_SOURCE_DIR=/home/flynn/sources/for_casadi/acados
export LD_LIBRARY_PATH=/home/flynn/sources/for_casadi/acados/lib:$LD_LIBRARY_PATH
```

However, when I take those directories and move them outside the acados installation path, I can't run them anymore. Here's the error (trying with pendulum_example):
```
[...]
gcc -fPIC -DACADOS_WITH_QPOASES -c acados_solver_pendulum_ode.c -I ../../../../include/blasfeo/include/ -I ../../../../include/hpipm/include/ \
-I ../../../../include -I ../../../../include/acados/ -I ../../../../include/qpOASES_e/
acados_solver_pendulum_ode.c:7:10: fatal error: acados/utils/print.h: No such file or directory
 #include "acados/utils/print.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Eventually I've found out that I need this one more line in bashrc:
```bash
export C_INCLUDE_PATH=/home/flynn/sources/for_casadi/acados/include:/home/flynn/sources/for_casadi/acados/include/blasfeo/include:/home/flynn/sources/for_casadi/acados/include/hpipm/include:/home/flynn/sources/for_casadi/acados/include/acados:/home/flynn/sources/for_casadi/acados/include/qpOASES_e:$C_INCLUDE_PATH
```

But I still can't compile successfully, encountering a linking error:
```
[...]
gcc -fPIC -DACADOS_WITH_QPOASES -shared -o libacados_solver_pendulum_ode.so  main_pendulum_ode.o pendulum_ode_model/pendulum_ode_expl_ode_fun.o pendulum_ode_model/pendulum_ode_expl_vde_forw.o lin_res_r_cost/lin_res_r_cost.c  lin_res_r_e_cost/lin_res_r_e_cost.c  acados_solver_pendulum_ode.o -L ../../../../lib \
-L ../../../../lib \
-lacados -lhpipm -lblasfeo -lqpOASES_e -lm \

/usr/bin/ld: cannot find -lacados
/usr/bin/ld: cannot find -lhpipm
/usr/bin/ld: cannot find -lblasfeo
/usr/bin/ld: cannot find -lqpOASES_e
collect2: error: ld returned 1 exit status
```

So I've tried to symbolic-link those libraries in a system directory:
```bash
cd lib
sudo ln -s `pwd`/*.so /usr/local/lib
```
and I've finally make everything run!

Using Ubuntu 18.04 with CasADi 3.5

zanellia, 2019-11-25:
Hey @Maik93, have you changed 'ocp.acados_include_path' and 'ocp.acados_lib_path' in the generating Python code? These attributes affect directly the generated Makefile which, in your case, seems to be referring to the default relative path. Please let me know if this fixes your issue. I have added an explicit mention to this in the README.md (https://github.com/acados/acados/commit/7b3a7d5668b13731340bdfd07c79a9f8e51b7db4).

Maik93, 2019-11-25:
Cool, it seems to work correctly!

Further improvements on code readability can be done writing this:
```python
acados_source_path = os.environ['ACADOS_SOURCE_DIR']
ocp.acados_include_path = acados_source_path + '/include'
ocp.acados_lib_path = acados_source_path + '/lib'
```

having only this two rows in bashrc:
```bash
export ACADOS_SOURCE_DIR=<path_to_acados_source>
export LD_LIBRARY_PATH=<path_to_acados_source>/lib:$LD_LIBRARY_PATH
```

In this way paths can be even defined inside the library and not by the user. What do you think about it?

closed 2020-01-10 by zanellia