acados/acados #519: acados inequality constraints error

ctoumieh, 2019-11-30:
Hi,

When adding an inequality `acados_constraint`, I am getting the following error:

```
File "generate_c_code.py", line 178, in <module>
    acados_solver = generate_solver(model, ra, json_file = 'acados_ocp.json')
  File "/home/charbel/.local/lib/python3.6/site-packages/acados_template/generate_solver.py", line 21, in generate_solver
    json.dump(ocp_nlp, f, default=np_array_to_list)
  File "/usr/lib/python3.6/json/__init__.py", line 179, in dump
    for chunk in iterable:
  File "/usr/lib/python3.6/json/encoder.py", line 430, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/usr/lib/python3.6/json/encoder.py", line 404, in _iterencode_dict
    yield from chunks
  File "/usr/lib/python3.6/json/encoder.py", line 437, in _iterencode
    o = _default(o)
  File "/home/charbel/.local/lib/python3.6/site-packages/acados_template/acados_ocp_nlp.py", line 1052, in np_array_to_list
    return np_array.tolist()
AttributeError: 'acados_constraint' object has no attribute 'tolist'
```
Is this something that you have yet to implement or is the issue on my end?

Cheers,
CT

giaf, 2019-12-01:
@zanellia this is for you I guess.

zanellia, 2019-12-05:
Hey @ctoumieh! Sorry for the loose response, it seems that you are adding a dynamic attribute to the `acados_ocp_nlp` object. If this is true, that's not the correct way of specifying constraints. Could you please provide more info?

ctoumieh, 2019-12-05:
Hi,
@zanellia thank you for your response! I sent you the code to your email (can't share in public since It is used for the development of an unpublished paper).

closed 2019-12-17 by ctoumieh