acados/acados #533: Assertion errors due to pointer alignment on 32-bit OS

bcbarbara, 2020-01-09:
Hello,

I am trying to run the exported C-code of `acados` - generated via its Python interface - on an [Odroid XU4](https://wiki.odroid.com/odroid-xu4/odroid-xu4), but due to the 32-bit Ubuntu OS I have installed (which is the only one supported), I am getting errors associated to the checks that pointers should be aligned to 8 bytes. 

[image]

Is there any possibility to port `acados` to 32-bit system?

zanellia, 2020-01-09:
Hey Barbara! Could you printf sizeof(blasfeo_dmat)? The memory should be aligned to 8 bytes by construction unless variables of certain types change in size (I guess integers and pointers are of dimension 4 bytes in your case).

bcbarbara, 2020-01-09:
Hey Andrea, I just checked the size of the struct blasfeo_dmat and it is 32.

zanellia, 2020-01-10:
Then the problem is likely to be related to assign functions in acados. The same issue has been discussed here https://github.com/acados/acados/issues/317 and a potential solution is proposed here 
https://github.com/acados/acados/pull/408. I can try to fix that PR.

giaf, 2020-01-10:
@zanellia there is no point in fixing that PR, it is outdated and it was causing issues with Travis, that's why it never got merged.

What we need to know to fix @bcbarbara 's issue is where (file + line of code) the assertion fails, and have a look at what is going on there.

zanellia, 2020-01-10:
@giaf I have read the conversation and understand the reason why the PR was not merged. However, it's 12 lines of additions and 4 deletions, I thought it might still be a good starting point. If you know how to fix the problem, please go ahead :)

zanellia, 2020-01-10:
@bcbarbara could you try to obtain a stack trace with gdb (see e.g. this https://senarvi.github.io/stack-trace-with-gdb/)? That's going to tell us where the assertion fails.

bcbarbara, 2020-01-10:
Sure, @zanellia. Come back with the results to you ASAP!

bcbarbara, 2020-01-10:
Sure, @zanellia. I'll come back with the results ASAP!

bcbarbara, 2020-01-10:
Hello guys, that's what I've got:

[image]

zanellia, 2020-01-10:
alright, so the (first) problem should be in `external_function_param_casadi_assign` at this line: https://github.com/acados/acados/blob/master/acados/utils/external_function_generic.c#L947 (the first call to `assign_and_advance_double_ptrs` should pass since the alignment is enforced explicitly). @giaf do we really need double pointers to be 8-byte aligned? If so, it might make sense to add a for loop with an explicit alignment at every iteration (that would do nothing on a 64-bit system).

tmmsartor, 2020-01-13:
@bcbarbara acados c examples are running on your platform?

FreyJo, 2020-04-24:
Issues with 32 Bit architectures are now tracked here:
https://github.com/acados/acados/issues/599
Sorry that we couldnt fix the issues in time..

closed 2020-04-24 by FreyJo