acados/acados #561: acados_template pypi

MaciejKucia, 2020-03-18:
At a brief glance, it seems that building python acados_template is independent to main acados codebase.

1. Would it be possible to split it to a separate repo?
2. Could wheel be published to pypi?

Thanks!

FreyJo, 2020-03-18:
Hi @MaciejKucia ,

1. That would definitely be possible.
I think it wouldnt make sense though.
Often when making a change in C, there has to be a corresponding change in the Python interface, e.g. when adding an option, modifying external function used inside a C module etc.

2. It would be nice to have acados as a pip package.
But I think it should also include binaries of the shared libraries, which would have to be compiled and uploaded there.
I never did this and with different HPIPM and BLASFEO targets, OSs etc, this doesnt seem easy to me.

If someone is willing to push for this it might happen though, feel free to contribute :wink:

MaciejKucia, 2020-03-24:
For now I have proposed following change: https://github.com/acados/acados/pull/567

closed 2020-03-24 by MaciejKucia