acados/acados #565: Matlab: Avoid `source env.sh`

FreyJo, 2020-03-23:
It was suggested in the forum to set the environment variables in the MATLAB/Octave rc file, such that 
`source env.sh` can be avoided.
https://discourse.acados.org/t/matlab-mex-more-elegant-way-to-setup-env-sh/62/4

I found there is a similar file in Octave
https://octave.org/doc/v4.2.1/Startup-Files.html

Should be tested across multiple OS, Matlab, Octave versions.

giaf, 2020-03-27:
I would leave both options.

Personally I like the env.sh because I like to have everything local to the project folder, for development it makes easier and safer when I have multiple versions of acados/hpipm/blasfeo on the system.

But I agree that for an end user, having to set this once forever in some config file may be more convenient. Instructions on alternatives to env.sh should definitely be added in the READMEs.

sandmaennchen, 2024-07-12:
The documentation now includes a remark on how to avoid `source env.sh`.

https://github.com/acados/acados/blob/master/docs/matlab_octave_interface/index.md#export-environment-variables

closed 2024-07-12 by sandmaennchen