acados/acados #589: Video results

Ilyaprok, 2020-04-16:
Hello, thank you very much for sharing this project. I am interested in NMPC, MHE in embedded,
particularly in STM32. Do you have any videos or any other results of yours NMPC? Maybe any comparasion videos with different controllers? Thank you:)

FreyJo, 2020-04-24:
Hi,

unfortunately, we dont have such a video available yet.
Note that acados does not support 32 Bit architectures at the moment, https://github.com/acados/acados/issues/599 .
However, if you could also run your NMPC application on a 64 bit one, please feel encouraged and let us know if you have further questions.

Cheers!

FreyJo, 2020-05-14:
A video result of a race car application is now available:
https://www.youtube.com/watch?v=1JDBQXVrZbo
https://github.com/acados/acados/commit/654088a48f296c870a4eb8eadac6dcb11b76c062

closed 2020-05-14 by FreyJo