acados/acados #599: Support 32 Bit Architectures

FreyJo, 2020-04-24:
I put a bit of work into this, see #597 but there are still some issues left.
For now, acados is not fully supported on 32 Bit architectures, mainly because of memory alignment issues.

In order to be able to make such a statement, the following should be done on a 32 Bit machine:
- [ ] make C tests / examples work
- [ ] make the Python tests work
- [ ] ideally have some CI to check some form of 32 Bit compatibility.

I hope this Guideline will be helpful for future developers/maintainers:
https://github.com/acados/acados/blob/master/docs/memory_management.md

giaf, 2021-04-08:
A step in the direction of fixing 32-bit support would be to edit these asserts
https://github.com/acados/acados/blob/master/acados/utils/mem.c#L95
https://github.com/acados/acados/blob/master/acados/utils/mem.c#L108
https://github.com/acados/acados/blob/master/acados/utils/mem.c#L121
https://github.com/acados/acados/blob/master/acados/utils/mem.c#L134
https://github.com/acados/acados/blob/master/acados/utils/mem.c#L147
such that they check for 8-byte alignment on 64-bit systems and 4-byte alignment on 32-bit systems.
(BLASFEO structs only contain int and pointers, so they are also good with the alignment for the pointers).

FreyJo, 2021-05-21:
Just to update this issue:
This PR applied the changes suggested in the comment above: https://github.com/acados/acados/pull/706

> A step in the direction of fixing 32-bit support would be to edit these asserts
> https://github.com/acados/acados/blob/master/acados/utils/mem.c#L95 [...]