acados/acados #606: Acados on conda-forge

pariterre, 2020-05-20:
Hi there!

I am writting a biomechanics frame work for optimal control called BiorbdOptim. It is based on CasADi and biorbd. I would be interested to support acados as well (we actually already working on it :) ). However, the installation process must be extra super easy for non programmers out there. 

I wanted to know if you guys would be interested that I maintain a built version of acados on conda-forge, so installing the last releases of acados would be as easy as typing the following command: `conda install acados -cconda-forge`. I already maintain CasADi and I think it would make sense to have acados on conda-forge as well. 

If you think that is a good idea but prefer to maintain it yourself, I can assist you on setting up things (I am currently maintaining probably 10 packages already on conda-forge)

Thanks! See you soon, and say hi to the team for me :)

FreyJo, 2020-05-20:
Hi @pariterre ,

Good to hear from you, that sounds great!
I did not use conda-forge myself yet, but seems like a cool project.
It would definitely be great to have this install option available.

I think, no one else is currently eager to set it up, so please feel encouraged to do so.
If you need help, just ask here or in the forum.
I saw that they are the projects are built for Linux, Mac, Windows.
About the Blasfeo/HPIPM targets, are you just going to compile them with target `GENERIC`, to make it work on every architecture and users will have to install manually if they want to use another (more efficient) target?

pariterre, 2020-05-20:
Yep, the GENERIC sounds the best.
I'll have a try within the next days!

wenlzhang, 2020-10-06:
It would be really great to be able to install `acados` from `conda-forge`. Is there any update on this?

pariterre, 2020-10-06:
@wenlzhang 
The main problem are the dependencies of `acados`. In order to have one, you have to have all the previous one as well. I did not have the courage to dig into this...

wenlzhang, 2020-10-06:
@pariterre That's good to know anyway. I was attracted by the description "super easy for non programmers out there"😀, as, for me, it was really complicated to debug and install the tool. Hopeful it would become easier to install and configure the tool soon.

pariterre, 2020-10-06:
@wenlzhang 
In https://github.com/pyomeca/bioptim, I created an initialization script that download and install everything properly (by automatically changing some piece of code of ACADOS that fails when installed). You may want to have a look at the script :)

wenlzhang, 2020-10-06:
@pariterre Many thanks for the script. I assume you meant the following one:
https://github.com/pyomeca/bioptim/blob/master/external/acados_install.sh

I will test it when I have time.

pariterre, 2020-10-06:
That is correct! 
If you are using conda, it should install acados in the current loaded environment (or a specified one in the parameters), and will choose GENERIC (unless specified otherwise in the parameters!)

FreyJo, 2020-10-07:
Hi @pariterre 

I had a look at your installation script and wonder:
Why do you have to replace the `python_requires` and the `CasADi` dependency?
Would a more loose version requirement, like this https://github.com/acados/acados/commit/d31f94b4c0abee818ade93c18721218996b2a740 fix those potential issues.
I remember Python >= 3.5 is a strict requirement for the acados Python interface.

Greets

pariterre, 2020-10-07:
@FreyJo 
Hi there! :)
That's right, when I made the script, it was limited to >=3.5 to <3.7, while bioptim necesitate 3.7 at least :)

FreyJo, 2020-10-16:
@pariterre it seems the conda-forge `acados` project is on hold now.
I suggest merge https://github.com/acados/acados/pull/608 and close this issue.
Or do you think there is more to come?

pariterre, 2020-10-21:
There is definitely more to come, but not in the short terms and it is more related to the work I am doing on bioptim, so no need to keep the issue open here. When I'll work on this again I can open the issue again if it is okay for you!

closed 2024-05-06 by sandmaennchen