acados/acados #609: Error while executing minimal_example_ocp.m

DennisWeir, 2020-05-28:
Hi there,

I installed acados for Windows today and tried to execute the minimal_example_ocp.m according to the steps listed on the website.
First of all, while executing acados_env_variables_windows.m, I noticed the external/casadi-matlab folder was missing, so I downloaded casadi and copied it into that folder.
Now I get the following error message:

```
Error using mbuild (line 166)
Unable to complete successfully.
   Creating library C:\acados\examples\acados_matlab_octave\getting_started\libpendulum.lib and object
   C:\acados\examples\acados_matlab_octave\getting_started\libpendulum.exp
LINK : fatal error LNK1104: cannot open file 'C:\acados\examples\acados_matlab_octave\getting_started\libpendulum.lib'

Error in ocp_generate_casadi_ext_fun (line 170)
    mbuild(c_files_path{:}, '-output', lib_name, 'CFLAGS="$CFLAGS"', 'LDTYPE="-shared"', ['LDEXT=', ldext]);

Error in acados_ocp (line 141)
                ocp_generate_casadi_ext_fun(obj.model_struct, obj.opts_struct);

Error in minimal_example_ocp (line 105)
ocp = acados_ocp(ocp_model, ocp_opts);
```

Interestingly, right before that error occurs, MEX changes compilers from MinGW64 to Visual C++ 2013.

Does anyone know how to fix this? Many thanks in advance,

Dennis

FreyJo, 2020-05-28:
Hi Dennis,

I guess you followed:
https://docs.acados.org/installation/#windows-for-use-with-matlab

> First of all, while executing acados_env_variables_windows.m, I noticed the external/casadi-matlab folder was missing, so I downloaded casadi and copied it into that folder.

There would have been the option to set it up automatically, but it seems you did it right manually.

acados for Matlab on Windows only supports MinGW for now.
Did you explicitly select MinGW using both `mex` and `mbuild`?

Best,
Jonathan

DennisWeir, 2020-05-28:
Hi Jonathan,

that solved it... I set mex but forgot to set mbuild as well.
It's running now, thank you very much!

Regards,
Dennis

closed 2020-05-28 by DennisWeir