acados/acados #613: External Model Example

ZhaoxuanZhu, 2020-06-17:
Hello,
I am trying to run the examples located at `acados_root/examples/acados_python/external_model`, and I got some cmake error when I try to build the code inside the folder. In the "modules" folder, I had to make changes to the part from:
``find_library(CASADI_LIBRARY
    NAMES casadi
    HINTS ${CASADI_INCLUDE_DIR}/../lib $ENV{CASADI_PREFIX}/lib)
``
to
``find_library(CASADI_LIBRARY
    NAMES casadi
    HINTS ${CASADI_INCLUDE_DIR}/../.. $ENV{CASADI_PREFIX}/lib)
``
After this, I can cmake it with the mesage that the path and the library are found. However, I am still getting the error when I `make` it, and the error message goes:
``
fatal error: casadi/casadi.hpp: No such file or directory
    2 | #include <casadi/casadi.hpp>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/external_ode_casadi.dir/build.make:63: CMakeFiles/external_ode_casadi.dir/src/external_ode_casadi.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/external_ode_casadi.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
``
Is this the issue of my environment variable or the specific version?
Thank you!

FreyJo, 2020-06-18:
Hi,
I think you need to compile CasADi yourself to run this example.

ZhaoxuanZhu, 2020-06-21:
Hey @FreyJo ,
Thanks for the reply and the fix does the job.

closed 2020-06-22 by FreyJo