acados/acados #617: Jbu/ubu/lbu missing check

DennisWeir, 2020-07-06:
When not providing Jbu,lbu,ubu at all the code compiles and runs, but does not calculate u.
When running the MPC as an S-Function, this creates problems with following blocks.
When one of the three is provided, the check at compile time will make sure that all three are set.

FreyJo, 2020-07-31:
Hi @DennisWeir ,

there are checks on Jbu, lbu, ubu:
https://github.com/acados/acados/blob/eb94a004e9f7aac12e191a1b0a8fcf65db930146/interfaces/acados_matlab_octave/detect_dims_ocp.m#L136-L146
What do you mean with it does not calculate u?

Although it is good practice, it is not mandatory to have bounds on u, which is why they are not required.
For example I just tried the following adaption of the minimal example https://github.com/acados/acados/blob/eb94a004e9f7aac12e191a1b0a8fcf65db930146/examples/acados_matlab_octave/getting_started/minimal_example_ocp.m#L1:

Which has no Jbu, lbu, ubu, but also doent need to have bounds on the controls at all:

```
clear all

model_path = fullfile(pwd,'..','pendulum_on_cart_model');
addpath(model_path)

check_acados_requirements()

%% discretization
N = 20;
T = 1; % time horizon length
x0 = [0; pi; 0; 0];

nlp_solver = 'sqp'; % sqp, sqp_rti
qp_solver = 'partial_condensing_hpipm';
    % full_condensing_hpipm, partial_condensing_hpipm, full_condensing_qpoases
qp_solver_cond_N = 5; % for partial condensing
% integrator type
sim_method = 'erk'; % erk, irk, irk_gnsf

%% model dynamics
model = pendulum_on_cart_model;
nx = model.nx;
nu = model.nu;

%% model to create the solver
ocp_model = acados_ocp_model();
model_name = 'pendulum';

%% acados ocp model
ocp_model.set('name', model_name);
ocp_model.set('T', T);

% symbolics
ocp_model.set('sym_x', model.sym_x);
ocp_model.set('sym_u', model.sym_u);
ocp_model.set('sym_xdot', model.sym_xdot);

% cost
ocp_model.set('cost_expr_ext_cost', model.expr_ext_cost);
ocp_model.set('cost_expr_ext_cost_e', model.expr_ext_cost_e);

% dynamics
if (strcmp(sim_method, 'erk'))
    ocp_model.set('dyn_type', 'explicit');
    ocp_model.set('dyn_expr_f', model.expr_f_expl);
else % irk irk_gnsf
    ocp_model.set('dyn_type', 'implicit');
    ocp_model.set('dyn_expr_f', model.expr_f_impl);
end

% constraints
ocp_model.set('constr_type', 'bgh');
% ocp_model.set('constr_expr_h', model.expr_h);
% U_max = 80;
% ocp_model.set('constr_lh', -U_max); % lower bound on h
% ocp_model.set('constr_uh', U_max);  % upper bound on h

ocp_model.set('constr_x0', x0);
% ... see ocp_model.model_struct to see what other fields can be set

%% acados ocp set opts
ocp_opts = acados_ocp_opts();
ocp_opts.set('param_scheme_N', N);
ocp_opts.set('nlp_solver', nlp_solver);
ocp_opts.set('sim_method', sim_method);
ocp_opts.set('qp_solver', qp_solver);
ocp_opts.set('qp_solver_cond_N', qp_solver_cond_N);
% ... see ocp_opts.opts_struct to see what other fields can be set

%% create ocp solver
ocp = acados_ocp(ocp_model, ocp_opts);

x_traj_init = zeros(nx, N+1);
u_traj_init = zeros(nu, N);

%% call ocp solver
% update initial state
ocp.set('constr_x0', x0);

% set trajectory initialization
ocp.set('init_x', x_traj_init);
ocp.set('init_u', u_traj_init);
ocp.set('init_pi', zeros(nx, N))

% change values for specific shooting node using:
%   ocp.set('field', value, optional: stage_index)
ocp.set('constr_lbx', x0, 0)

% solve
ocp.solve();
% get solution
utraj = ocp.get('u');
xtraj = ocp.get('x');

status = ocp.get('status'); % 0 - success
ocp.print('stat')

%% Plots
figure; hold on;
States = {'p', 'theta', 'v', 'dtheta'};
for i=1:length(States)
    subplot( length(States), 1, i);
    plot(xtraj(i,:)); grid on;
    legend(States{i});
end

figure
stairs(utraj')
grid on

%% go embedded
% to generate templated C code
% ocp.generate_c_code;
```

I didnt test that from Simulink.
If there is a specific problem there, please give a more precise description.

DennisWeir, 2020-08-04:
Hi Jonathan,

when I didn't pass any of the three, in Simulink there was no output value.
Not even a scope was able to record a signal for several time steps. Until the delayed (nonexistent) signal reached the (Amesim) simulation model which then triggered an exit.
Sorry for the delayed reply and not being able to test it again currently, but I'm two weeks short of handing in my master thesis, so lots of other things to do..

Best wishes
Dennis

closed 2020-10-07 by FreyJo