acados/acados #643: SCQP example not runable

syangav, 2020-10-07:
Hi, 

Thanks for your contribution on this wonderful framework! 

I am trying to run the SCQP example in examples/c/ by uncommenting this line [1](https://github.com/acados/acados/blob/4dd0f775450af9d47615f985066c923c3d2b51c2/examples/c/CMakeLists.txt#L249) However, several errors bumped out, namely BGHP no longer exists but it appears in [2](https://github.com/acados/acados/blob/4dd0f775450af9d47615f985066c923c3d2b51c2/examples/c/pendulum_scqp.cpp#L93), nl_constr_h_fun_jac shall only exist in ocp_nlp_constraints_bgh_model but appears in [3](https://github.com/acados/acados/blob/4dd0f775450af9d47615f985066c923c3d2b51c2/examples/c/pendulum_scqp.cpp#L213) and the p in [4](https://github.com/acados/acados/blob/4dd0f775450af9d47615f985066c923c3d2b51c2/examples/c/pendulum_scqp.cpp#L214) is undefined. It seems that the pull request #506 did not fix all the problems. Any suggestions on workaround or solution? 

Thanks in advance for your time and attention. 

Best, 
Yang

syangav, 2020-10-07:
May I also ask how are the files constraint.h, constraint.c, position.h, position.c in examples/c/pendulum_model generated via CasADi? I can see the pendulum_ode_* files are generated by the MATLAB file generate_C_model.m but for constraints, there are no such files.

FreyJo, 2020-10-07:
Hi Yang,

This example is broken as discussed in the PR you mentioned:
https://github.com/acados/acados/pull/506/files#r341644335
I am going to move it away from the examples folder.

If you want to check out an example for SCQP (sequential convex quadratic programming), i.e. that uses the convex over nonlinear constraint module BGP, I recommend these two:
https://github.com/acados/acados/blob/master/examples/acados_python/pmsm_example/generate_c_code.py
https://github.com/acados/acados/blob/master/examples/acados_python/rsm_example/generate_c_code.py

These are Python examples, but the Python interface is based on templates which basically generate a `C` example from it.
Just since you were looking in `examples/c`.

Best,
Jonathan

FreyJo, 2020-10-07:
> May I also ask how are the files constraint.h, constraint.c, position.h, position.c in examples/c/pendulum_model generated via CasADi? I can see the pendulum_ode_* files are generated by the MATLAB file generate_C_model.m but for constraints, there are no such files.

You should be also able to find that out via the Python examples, respectively the Python interface.
That's also why I think these examples are much better to understand what is going on.

syangav, 2020-10-13:
Thank you very much for your fast reply. I think that fixed my question. Closing now.

closed 2020-10-13 by syangav