acados/acados #646: OpenMP-Problem while running  minimal_example_ocp in MATLAB/Windows

RicusH, 2020-10-14:
Hi,
first of all thanks for the great work!

I got an error running the "minimal_example_ocp.m" after the installation following "https://docs.acados.org/installation/index.html#windows-for-use-with-matlab" concerning the comilation of "ocp_create.c" etc. and some omp-Functionalities. 

I fixed it by switching -D ACADOS_WITH_OPENMP=OFF for the build. 

I don't know if this is a bug of such or just an user error but you might want to point to setting this option in the installation-guide.

Thanks and all the best!

Ricus

FreyJo, 2020-10-14:
Hi Ricus,

thanks for reporting that.
Someone already mentioned this Problem here.
https://discourse.acados.org/t/acados-pre-release-v0-1-3/270/3

I dont know what the issue is, since CMake seems to find OpenMP and we set the flags.
Maybe you can post the error message.

I consider turning `ACADOS_WITH_OPENMP=OFF` by default, if there is no solution soon.

Best,
Jonathan

RicusH, 2020-10-14:
Hi Jonathan,

sorry, I missed that post in https://discourse.acados.org/t/acados-pre-release-v0-1-3/270 . 

The error message was the same (undefined reference to 'omp_get_num_threads') . 

All the best
Ricus

FreyJo, 2020-10-14:
I added the flag in the installation instructions. Might take a while before I look into this..
https://github.com/acados/acados/commit/3b5e1504e0127dded60fcb0e7f07717f09dab566

giaf, 2020-10-14:
@FreyJo I would always set OpenMP off by default in any case and in all architectures, also without bugs involved.
And let the user turn it on if s/he needs and wants to try out if this gives some extra speed.

IMO this also makes sense in regards of embedded-friendliness and reliable timings (which are much less in case of multi-threading involved).

FreyJo, 2020-10-14:
@giaf ok, that makes sense! I turned it off by default here:
https://github.com/acados/acados/pull/648

closed 2020-10-14 by FreyJo