acados/acados #647: Adding comments and instructions on examples/c?

syangav, 2020-10-14:
Hi, 

I am wondering whether you have the plan for finishing the examples in the doc [1](https://docs.acados.org/examples/index.html) or providing a really simple and readable C getting started example code for optimal nonlinear control problem.Unlike the Matlab and python ones, I have found the C examples hard to understand and lack of comment. I also noticed that the python API is well documented since every property is followed by some clarification [2](https://docs.acados.org/interfaces/index.html#module-acados_template.acados_ocp). Unfortunately, this is not the case for c API. Will you improve it? 

Best,
Yang

FreyJo, 2020-10-16:
Hi,

yes, it would be actually nice to have something on this docs page https://docs.acados.org/examples/index.html
I am not sure though which form would be the most useful, maybe some walkthrough on the Python example, plus maybe explain how the generated code can be used.
What do you think?

Regarding the `C` examples, I think there wont be much effort in the near future to polish them.
As mentioned in https://github.com/acados/acados/issues/643 , the Python examples basically generate a `C` example that is relatively readable.
Alternatively, you can look at the templates, the generated Code is based on, e.g. here https://github.com/acados/acados/blob/master/interfaces/acados_template/acados_template/c_templates_tera/acados_solver.in.c
This should make clear how the documented properties from Python are translated to the C API.

Best,
Jonathan

FreyJo, 2020-11-23:
Closing due to inactivity..

closed 2020-11-23 by FreyJo