acados/acados #650: Handling One-Sided Constraints

FreyJo, 2020-10-15:
The `acados` problem description is based on two sided constraints.
Adding one-sided constraints in this framework is nontrivial and can lead to ill conditioning, especially in the QP solver.
The most user-friendly way would be to allow the user to set them to `Inf` and convert them internally.

- for HPIPM, recently constraint `mask` was added to deal with one sided constraints.
They can be changed after creation and dont change the memory needed.
https://github.com/giaf/hpipm/blob/b86f426a2de7bd8fdf35aa63c9f6b59146d8eae5/Changelog.txt#L67
https://github.com/giaf/hpipm/blob/b86f426a2de7bd8fdf35aa63c9f6b59146d8eae5/include/hpipm_d_ocp_qp.h#L116
- for other QP solvers, the plan would be to replace the `Inf` with some `ACADOS_INF` value that has to be defined and should be tested on some example

That is just a rough plan of what should be done, any comments welcome!

For now, we recommend to use not too large numbers for unbounded sides of the constraint to avoid ill-conditioning.

FreyJo, 2024-08-02:
Preliminary support for this is added in https://github.com/acados/acados/pull/1174
Follow-ups include:
- more testing, feedback from the community
- Support in QP solvers other than HPIPM
   - [ ] DAQP
   - [ ] qpOASES
   - [ ] qpDUNES
   - [ ] OSQP

sandmaennchen, 2024-08-09:
Further point to consider:

- [ ] Multiplier associated with masked bounds should always be zero. In particular, this needs to be ensured whenever multipliers are initialized and whenever the bounds are updated (potentially masking previously active constraints)

sandmaennchen, 2024-09-04:
I added a note in the updated problem description, https://github.com/acados/acados/pull/1221, which should be updated when one-sided constraints are fully supported.