acados/acados #651: Linker problem after building acados with multiple qp solvers

asparc, 2020-10-16:
I'm experiencing linker issues on Windows 10 + Matlab2017b with acados compiled with multiple qp solvers when running the minimal example in `acados_matlab_octave\getting\started\minimal_example_ocp.m`.

More specifically: 
I compiled acados with qpOASES and OSQP, by turning on their flags in the CMakeLists.txt. When I set the default qp solver (`partial_condensing_hpipm`), `minimal_example_ocp.m` fails because MinGW64 complains about undefined references to qpOASES and OSQP, while these solvers shouldn't be used.

```
Building with 'MinGW64 Compiler (C)'.
C:\ProgramData\MATLAB\SupportPackages\R2017b\3P.instrset\mingw_w64.instrset\bin\gcc -c -DUSE_MEX_CMD   -m64 -DMATLAB_MEX_FILE  -I"C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\.." -I"C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\interfaces" -I"C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\external" -I"C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\external\blasfeo\include" -I"C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\external\hpipm\include"  -I"C:\Program Files\MATLAB\R2017b/extern/include" -I"C:\Program Files\MATLAB\R2017b/simulink/include" -I"C:\Program Files\MATLAB\R2017b/extern\lib\win64\mingw64" -fexceptions -fno-omit-frame-pointer -std=c99 -fopenmp -O -DNDEBUG "C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\interfaces\acados_matlab_octave\ocp_create.c" -o C:\Users\GRUIJT~1\AppData\Local\Temp\mex_136896563278865_18692\ocp_create.obj
C:\ProgramData\MATLAB\SupportPackages\R2017b\3P.instrset\mingw_w64.instrset\bin\gcc -c -DUSE_MEX_CMD   -m64 -DMATLAB_MEX_FILE  -I"C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\.." -I"C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\interfaces" -I"C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\external" -I"C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\external\blasfeo\include" -I"C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\external\hpipm\include"  -I"C:\Program Files\MATLAB\R2017b/extern/include" -I"C:\Program Files\MATLAB\R2017b/simulink/include" -I"C:\Program Files\MATLAB\R2017b/extern\lib\win64\mingw64" -fexceptions -fno-omit-frame-pointer -std=c99 -fopenmp -O -DNDEBUG "C:\Program Files\MATLAB\R2017b\extern\version\c_mexapi_version.c" -o C:\Users\GRUIJT~1\AppData\Local\Temp\mex_136896563278865_18692\c_mexapi_version.obj
C:\ProgramData\MATLAB\SupportPackages\R2017b\3P.instrset\mingw_w64.instrset\bin\gcc -m64 -Wl,--no-undefined -fopenmp -shared -static -s -Wl,"C:\Program Files\MATLAB\R2017b/extern/lib/win64/mingw64/exportsmexfileversion.def" C:\Users\GRUIJT~1\AppData\Local\Temp\mex_136896563278865_18692\ocp_create.obj C:\Users\GRUIJT~1\AppData\Local\Temp\mex_136896563278865_18692\c_mexapi_version.obj   -llibacados -llibhpipm -llibblasfeo  -LC:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\lib   -L"C:\Program Files\MATLAB\R2017b\extern\lib\win64\mingw64" -llibmx -llibmex -llibmat -lm -llibmwlapack -llibmwblas -o C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\getting_started\build\ocp_create.mexw64
Error using mex
C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\lib\libacados.lib(dense_qp_qpoases.c.obj):dense_qp_qpoases.c:(.text+0x186):
undefined reference to `QProblem_calculateMemorySize'
C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\lib\libacados.lib(ocp_qp_osqp.c.obj):ocp_qp_osqp.c:(.text+0xac):
undefined reference to `osqp_set_default_settings'

--> many more undefined reference errors for qpOASES and OSQP, just like the two I handpicked above

Error in ocp_compile_interface (line 198)
            mex('-v', mex_flags, FLAGS, LDFLAGS, acados_include, acados_interfaces_include, external_include, blasfeo_include, hpipm_include,...
```

I can fix this problem by adding `-losqp` and `-lqpOASES_e` to each of the `mex()` commands around line 160-170 in `ocp_compile_interface.m`:
```
mex('-v', mex_flags, FLAGS, LDFLAGS, acados_include, acados_interfaces_include, external_include, blasfeo_include, hpipm_include, acados_lib_path, '-lacados', '-lhpipm', '-lblasfeo', '-losqp', '-lqpOASES_e', mex_files{ii}, '-outdir', opts.output_dir)
```

However, this is just a quick patch, especially as it requires users to manually change all `mex()` commands in `ocp_compile_interface.m` when different solvers are selected.

Note that this linker error only occurs in the third gcc call that the `mex()` command invokes, i.e. where it actually creates the `.mexw64` file:
```
C:\ProgramData\MATLAB\SupportPackages\R2017b\3P.instrset\mingw_w64.instrset\bin\gcc -m64 -Wl,--no-undefined -fopenmp -shared -static -s -Wl,"C:\Program Files\MATLAB\R2017b/extern/lib/win64/mingw64/exportsmexfileversion.def" C:\Users\GRUIJT~1\AppData\Local\Temp\mex_136896563278865_18692\ocp_create.obj C:\Users\GRUIJT~1\AppData\Local\Temp\mex_136896563278865_18692\c_mexapi_version.obj   -llibacados -llibhpipm -llibblasfeo  -LC:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\..\..\lib   -L"C:\Program Files\MATLAB\R2017b\extern\lib\win64\mingw64" -llibmx -llibmex -llibmat -lm -llibmwlapack -llibmwblas -o C:\Users\gruijthuijsenc\Drive\Documents\MATLAB\acados\examples\acados_matlab_octave\getting_started\build\ocp_create.mexw64
```
This compilation step seems to be very Matlab specific and I don't how to interpret this.

FreyJo, 2020-10-16:
Hi,
thanks for reporting this!
I am working on adding support for QP solvers through the high-level interfaces (Matlab and Python) here: https://github.com/acados/acados/pull/644
I will also try to look at the case that the `acados` library was compiled with multiple additional QP solvers, given your issue.

asparc, 2020-10-16:
Nice that you're working on that now!

I came across a couple of other issues that prevented me from running the minimal_example.m, but I managed to fix these issues myself. I created a pull request for my fixes.

giaf, 2020-10-26:
The observed behavior is not surprising, it is actually the expected behavior given what happens. Let me explain.
When acados is compiled with e.g. ACADOS_WITH_QPOASES (or any other optional solver), the corresponding wrapper to the qpOASES solver is compiled too; this wrapper clearly contains calls to qpOASES functions, which are in the qpOASES library, and hence in the acados wrapper there are undefined symbols.
When acados is compiled without ACADOS_WITH_QPOASES, the wrapper to qpOASES is not compiled nor included in the acados library, hence there are no undefined symbols to qpOASES functions.
The mex interface makes use of shared libraries. When using shared libraries, all undefined symbols are supposed to be resolved in the provided linking libraries at linking time. Thus, if the corresponding qpOASES library is not provided, the linker will complain about undefined symbols.

One solution could be to create at compile time a header file in acados, which holds e.g. the actual solvers acados library has been compiled for.
Then the linking libraries in case of frameworks using the shared acados library (such e.g. mex in matlab/octave) should be selected based on this header file, rather than on the specific solver which it is intended to be used in the example.

asparc, 2020-10-26:
Thanks the details! It makes a lot of sense what you say. 

The current, manual way of doing it is possible of course, but creates quite some complexity for users of a high-level interface. You have to specify in a multitude of places which libraries to link, e.g. in 
- acodos_compile_interface.m
- ocp_set_ext_fun.m
and also in autogenerated make files for autogenerated C code, like make_main_mex_pendulum_32.m, if I understand correctly what this file (located in the c_generated_code folder with the minimal_examples) represents. 

It would be useful if there was just one centralized place to specify these linking libs. This information can indeed come from a autogenerated header file, but IMO any other manually editable settings file would already be a big step forward.

asparc, 2020-10-27:
Did you intentionally close this? Would be fine by me, just wondering :-)

FreyJo, 2020-10-27:
No sorry, must have been a missclick :see_no_evil:

asparc, 2021-04-06:
I picked up this issue again, as I was making the Simulink S-function compilation work on Windows (which was still unsupported) and I wanted to make the compilation process to support multiple solvers. To tell the mex/mingw32-make commands which libraries to link to, I generate a .json file at the end of CMakeLists.txt that stores which solvers/options acados is built with, like so:
```cmake
# Write libraries to be linked
if(${ACADOS_WITH_OPENMP}) 
    set(WRITE_OPENMP -fopenmp)
endif()
if(${ACADOS_WITH_QPOASES}) 
    set(WRITE_QPOASES -lqpOASES_e)
endif()
if(${ACADOS_WITH_HPMPC}) 
    set(WRITE_HPMPC -lhpmpc)
endif()
if(${ACADOS_WITH_OSQP}) 
    set(WRITE_OSQP -losqp)
endif()
file(WRITE lib/link_libs.json {\n)
file(APPEND lib/link_libs.json \t\"openmp\":\ \"${WRITE_OPENMP}\",\n)
file(APPEND lib/link_libs.json \t\"qpoases\":\ \"${WRITE_QPOASES}\",\n)
file(APPEND lib/link_libs.json \t\"hpmpc\":\ \"${WRITE_HPMPC}\",\n)
file(APPEND lib/link_libs.json \t\"osqp\":\ \"${WRITE_OSQP}\"\n) # no final comma!
file(APPEND lib/link_libs.json })
```

If you think such hack in the CMakeLists is acceptable, I could create a pull request with all my changes to generate S-functions on Windows.

FreyJo, 2024-05-10:
This has been fixed in https://github.com/acados/acados/pull/712

closed 2024-05-10 by FreyJo