acados/acados #667: QP solver erro status 3

alexglzg, 2020-11-21:
Hello, 
I am using the following code for qp_solver and I am getting status error #3. 
I understand error code 3 means: minimum step size in QP solver reached. 
I tried different combinations of N (# of intervals in the horizon) and Tf (prediction horizon), but so far nothing has worked. 
```
## set QP solver
#ra.solver_options.qp_solver = 'FULL_CONDENSING_QPOASES'
ra.solver_options.qp_solver = 'PARTIAL_CONDENSING_HPIPM'
ra.solver_options.hessian_approx = 'GAUSS_NEWTON'
ra.solver_options.integrator_type = 'ERK'

# set prediction horizon
ra.solver_options.tf = Tf
ra.solver_options.nlp_solver_type = 'SQP_RTI'
#ra.solver_options.nlp_solver_type = 'SQP'
```

Can you give me any advise on how to solve this problem?

FreyJo, 2020-11-23:
The part of the code that you posted is most likely not the reason for your error.

> I understand error code 3 means: minimum step size in QP solver reached.

Correct. This means that the problem is likely to be infeasible.
Could this be the case?

In general, these kind of questions are better suited for the forum https://discourse.acados.org

alexglzg, 2020-11-23:
Hi @FreyJo 

After some time debugging, the acados_solver status was 4, not 3 as the QP solver was displaying. I agree the problem is likely to be infeasible, I think I am starting with a singularity or infeasible state. I changed the initial state x0, and the error stop appearing, the code is now working as long as I avoid an x0 of all-zeros.

Thanks for your help! I will use the forum instead, in case of future questions.

closed 2020-11-23 by alexglzg