acados/acados #677: Problem in backtracking linesearch?

jgillis, 2020-11-30:
The following snippet performs 19 iterations of SQP, while only a single one is expected (because it is a QP)

```python
#
# Copyright 2019 Gianluca Frison, Dimitris Kouzoupis, Robin Verschueren,
# Andrea Zanelli, Niels van Duijkeren, Jonathan Frey, Tommaso Sartor,
# Branimir Novoselnik, Rien Quirynen, Rezart Qelibari, Dang Doan,
# Jonas Koenemann, Yutao Chen, Tobias Schöls, Jonas Schlagenhauf, Moritz Diehl
#
# This file is part of acados.
#
# The 2-Clause BSD License
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are met:
#
# 1. Redistributions of source code must retain the above copyright notice,
# this list of conditions and the following disclaimer.
#
# 2. Redistributions in binary form must reproduce the above copyright notice,
# this list of conditions and the following disclaimer in the documentation
# and/or other materials provided with the distribution.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
# AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
# IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
# ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
# LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
# CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
# SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
# INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
# CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
# ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
# POSSIBILITY OF SUCH DAMAGE.;
#

import sys
sys.path.insert(0, '../getting_started/common')

from acados_template import *
from pendulum_model import export_pendulum_ode_model
import numpy as np
import scipy.linalg
import argparse

# set to 'True' to generate test data
GENERATE_DATA = False

LOCAL_TEST = False
TEST_TOL = 1e-8

if LOCAL_TEST is True:
    COST_MODULE = 'LS'
    COST_MODULE_N = 'LS'
    SOLVER_TYPE = 'SQP_RTI'
    QP_SOLVER = 'FULL_CONDENSING_QPOASES'
    INTEGRATOR_TYPE = 'IRK'
    HESS_APPROX = 'GAUSS_NEWTON'
else:
    parser = argparse.ArgumentParser(description='test Python interface on pendulum example.')
    parser.add_argument('--COST_MODULE', dest='COST_MODULE',
                        default='LS',
                        help='COST_MODULE: linear least-squares (LS) or nonlinear \
                                least-squares (NLS) (default: LS), external (EXTERNAL)')

    parser.add_argument('--COST_MODULE_N', dest='COST_MODULE_N',
                        default='LS',
                        help='COST_MODULE_N: linear least-squares (LS) or nonlinear \
                                least-squares (NLS) (default: LS), external (EXTERNAL)')

[... 197 lines cut ...]

ocp_solver = AcadosOcpSolver(ocp, json_file = 'acados_ocp.json')
ocp_solver.options_set("globalization", "merit_backtracking")

# initialize solver
x_traj_init = np.transpose( np.vstack( [np.zeros((N+1,)), np.zeros((N+1,))]) )
for i in range(N+1):
    ocp_solver.set(i, "x", x_traj_init[i])

pi_init = np.ones((N, nx))
for i in range(N):
    ocp_solver.set(i, "pi", pi_init[i])

u_init = np.zeros((N, nu))
for i in range(N):
    ocp_solver.set(i, "u", u_init[i])

# solve ocp
simX = np.ndarray((N+1, nx))
simU = np.ndarray((N, nu))

status = ocp_solver.solve()

ocp_solver.print_statistics()
```

jgillis, 2020-11-30:
Merit function is zero, despite a clear violation of initial constraints:

```
i 0
0.00000e+00 5.00000e+00 0.00000e+00 -5.00000e+00 

0.00000e+00 0.00000e+00 0.00000e+00 0.00000e+00 

IN merit fun: ineq i 0, j 0 tmp_fun0.000000e+00, multiplier 0.000000e+00
IN merit fun: ineq i 0, j 1 tmp_fun5.000000e+00, multiplier 0.000000e+00
IN merit fun: ineq i 0, j 2 tmp_fun0.000000e+00, multiplier 0.000000e+00
IN merit fun: ineq i 0, j 3 tmp_fun-5.000000e+00, multiplier 0.000000e+00
```

https://github.com/acados/acados/blob/f7a50c9ba8dcb59c2d4357e14f43c9ea49230383/acados/ocp_nlp/ocp_nlp_common.c#L2190

FreyJo, 2020-12-01:
Hi,
thanks for reporting this.
The contribution of the constraints to the merit function is zero intially, because the inequality multipliers are initialized with zeros.
Initializing them to 1e1 or bigger, i.e. these lines:
```
lam_0 = 1e1
for i in range(N+1):
    if i == 0:
        lam_init = lam_0 * np.ones((2*2,1))
    elif i == N:
        lam_init = lam_0 * np.ones((2,1))
    else:
        lam_init = lam_0 * np.ones((0,))
    ocp_solver.set(i, "lam", lam_init)
```
makes the algorithm converge within one iteration as expected:
```
iter	res_stat	res_eq		res_ineq	res_comp	qp_stat	qp_iter
0	1.200000e+00	0.000000e+00	5.000000e+00	0.000000e+00	0	0
1	7.283063e-14	1.332268e-15	2.823093e-16	1.131268e-15	0	3
```

Note, that this is a preliminary implementation of the line search algorithm.
I'll try to look into the TODOs in `ocp_nlp_line_search` soon.

Maybe, we can also initialize the multipliers with nonzeros by default. Otherwise this is not really compatible with the initialization of the merit function weights as described in Leineweber1999.
I guess ones would be another natural choice.

closed 2022-03-14 by FreyJo