acados/acados #729: Error while testing minimal_example_ocp.py

MajdWardeh, 2021-06-13:
Hello,

I've installed Acados on my system (Ubuntu 18.04) using CMake as well as the Python interface.
I'm getting this error when I run the minimal_example_ocp.py:
```
Traceback (most recent call last):
  File "minimal_example_ocp.py", line 102, in <module>
    ocp_solver = AcadosOcpSolver(ocp, json_file = 'acados_ocp.json')
  File "/home/majd/Downloads/acados/interfaces/acados_template/acados_template/acados_ocp_solver.py", line 765, in __init__
    with open(json_path + '/simulink_default_opts.json', 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '~/Downloads/acados/interfaces/acados_template/acados_template/simulink_default_opts.json'
Exception ignored in: <bound method AcadosOcpSolver.__del__ of <acados_template.acados_ocp_solver.AcadosOcpSolver object at 0x7fe057c05f60>>
Traceback (most recent call last):
  File "/home/majd/Downloads/acados/interfaces/acados_template/acados_template/acados_ocp_solver.py", line 1454, in __del__
    model = self.acados_ocp.model
AttributeError: 'AcadosOcpSolver' object has no attribute 'acados_ocp'

```
I copied pendulum_model.py and utils.py to the ocp folder since python couldn't find them.
could you please help? 
Thank you,

FreyJo, 2021-06-14:
Do you have a clean git status and are on the latest master?
The file definitely is there in the repository..
https://github.com/acados/acados/blob/6926943f33776f0582c731b54056678a1d4b7a92/interfaces/acados_template/acados_template/simulink_default_opts.json

MajdWardeh, 2021-06-14:
Yes, I check it. The file exists.

MajdWardeh, 2021-06-14:
problem solved!
Just changed the '~' in the path to '/home/majd/'.

Thank you,

closed 2021-06-14 by MajdWardeh

CheeGY, 2021-06-22:
Hi @MajdWardeh ,

I am having a similar issue, but can't seem to resolve it. In which file did you change the '~' in the path?

Thanks!

MajdWardeh, 2021-06-22:
Hi CheeGY,

My mistake was that I used '~/Downloads/....' as the path for the `<acados_root> ` when I installed the Python interface in the following lines:
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:"<acados_root>/lib"
export ACADOS_SOURCE_DIR="<acados_root>"
```
Also, don't forget to add these two lines to your `.bashrc`/`.zshrc`.

> Hi @MajdWardeh ,
> 
> I am having a similar issue, but can't seem to resolve it. In which file did you change the '~' in the path?
> 
> Thanks!