acados/acados #738: Fixed iterations for QP solver

HaraldSchafer, 2021-07-22:
You can set the max iterations with "qp_iter_max". For real-time applications it also makes sense to have a minimum amount of operations. Is there a way to set the minimum iterations or somehow fix the amount of iterations?

Thanks

FreyJo, 2021-07-23:
I don't really think this makes sense.
The QPs are solved up to some tolerance, (`qp_tol*`). Since the QP formulation is based on a linearization of the NLP, one does not want to spend more time on a QP that is anyway just an approximation of the NLP.
If there is a lot of time left, one could just carry out another SQP iteration.

> or somehow fix the amount of iterations?

Especially, for active-set methods like `qpOASES`, one should not use non converged QP solutions, since the active-set is not found yet and the solution output is probably infeasible.

> For real-time applications it also makes sense to have a minimum amount of operations.

I am not really sure why this should be the case?
What is the issue when the OCP solver terminates early?

HaraldSchafer, 2021-07-23:
Hmmm I guess I am misunderstanding some things.

The issue I am having is the following. I am trying to port lateral and longitudinal planning in openpilot to acados. But it is a requirement for openpilot that 1 planning step runs in a quasi-fixed amount of time. Having the cpu usage  spike when the planning problem changes could cause the entire system to hang. Ideal behavior would be for it to run a fixed amount of time and return the best solution possible in that time.
https://github.com/commaai/openpilot/pull/21226

If that's not possible having some upper bound on cpu usage would be helpful too, but I'm not sure how to find that out either? Is there an easy way test the worst case in terms of compute requirements? 

Thanks for your help!

FreyJo, 2021-07-23:
I see, that requirement is more what I expected.

As I mentioned, I would not limit the number of QP iterations for an active-set method.
However, if you are using an interior-point method like `HPIPM`, the CPU time per QP iteration varies less, and it makes more sense to limit the number of QP iterations in this case.

In HPIPM there are different modes and options available.
By properly choosing options it is possible to make the IPM iteration time rather constant, at the expense of a less advanced algorithm.anyway
In addition to the modes (SPEED, BALANCE, ...), there is also an option to disable the conditional Mehrotra’s corrector step.
Details can be found in the HPIPM paper https://cdn.syscop.de/publications/Frison2020a.pdf
Note that not all of theses HPIPM options are interfaced as options in the acados C and Python interfaces yet.

I suggest to first try to get a good guess of how many HPIPM QP iterations you can carry out in the sampling time and set that as a maximum number of QP iterations.

HaraldSchafer, 2021-07-23:
Ah yes that was very helpful. I will switch everything to HPIPM and do some experimentation.

I can add some of the HPIPM options to the interfaces if they end up being necessary.

Thanks!!

closed 2022-03-14 by FreyJo