acados/acados #755: The accard can't run on my Windows 10

Faust-Wang, 2021-09-17:
I compiled successfully （mingw32-make.exe install） according to the tutorial （https://docs.acados.org/installation/index.html）
The MATLAB example, however, does not work. 
1.Error:  `MEX could not find the library 'acados' specified with the -l option.  
MEX has searched for files with one of the following names: 
 libacados.a
acados.lib
libacados.lib
Verify that the library name is correct.  If the library is not there  
In an existing path, use the -l option to specify the path.   `
I modify line 169 of 'ocp_compile_interface.m' (<'-lacados', '-lhpipm', '-lblasfeo',libs.osqp> to <'-Lacados', '-Lhpipm', '-Lblasfeo','-Losqp'>). 
Ok, this error is gone. But the second erroecame out and I couldn't fix it.
2. Error:
`--------------------------------------------------------------
Structure detection for path cost
Cost function is quadratic -> Reformulating as linear_ls cost.

reformulated cost term in linear least squares form with:
cost = 0.5 * || Vx * x + Vu * u + Vz * z - y_ref ||_W

Vx
     1     0     0     0
     0     1     0     0
     0     0     1     0
     0     0     0     1
     0     0     0     0

Vu
     0
     0
     0
     0
     1

Vz

W
   1.0e+03 *

    2.0000         0         0         0         0
         0    2.0000         0         0         0
         0         0    0.0000         0         0
         0         0         0    0.0000         0
         0         0         0         0    0.0000

y_ref
     0
     0
     0
     0
     0

y (symbolic)
[p, theta, v, dtheta, F]

NOTE: These numerical values can be updated online using the appropriate setters.
--------------------------------------------------------------
--------------------------------------------------------------
Structure detection for terminal cost term
Cost function is quadratic -> Reformulating as linear_ls cost.

reformulated cost term in linear least squares form with:
cost = 0.5 * || Vx * x + Vu * u + Vz * z - y_ref ||_W

Vx
     1     0     0     0
     0     1     0     0
     0     0     1     0
     0     0     0     1

Vu
[... 180 lines cut ...]
C:\Users\XXX\AppData\Local\Temp\mex_4329624523067918_22816\ocp_create.obj:ocp_create.c:(.text+0x6bf3): undefined reference to
`ocp_nlp_dims_set_opt_vars'
C:\Users\XXX\AppData\Local\Temp\mex_4329624523067918_22816\ocp_create.obj:ocp_create.c:(.text+0x6cfc): undefined reference to
`ocp_nlp_solver_opts_set_at_stage'
C:\Users\XXX\AppData\Local\Temp\mex_4329624523067918_22816\ocp_create.obj:ocp_create.c:(.text+0x6dc2): undefined reference to
`ocp_nlp_solver_opts_set_at_stage'
C:\Users\XXX\AppData\Local\Temp\mex_4329624523067918_22816\ocp_create.obj:ocp_create.c:(.text+0x6e79): undefined reference to
`ocp_nlp_solver_opts_set_at_stage'
C:\Users\XXX\AppData\Local\Temp\mex_4329624523067918_22816\ocp_create.obj:ocp_create.c:(.text+0x6f30): undefined reference to
`ocp_nlp_solver_opts_set_at_stage'
C:\Users\XXX\AppData\Local\Temp\mex_4329624523067918_22816\ocp_create.obj:ocp_create.c:(.text+0x6fe7): undefined reference to
`ocp_nlp_solver_opts_set_at_stage'
collect2.exe: error: ld returned 1 exit status

Error  ocp_compile_interface ( (line 169)
        mex(mex_flags, FLAGS, LDFLAGS, COMPDEFINES, COMPFLAGS, acados_include, acados_interfaces_include, external_include,
        blasfeo_include, hpipm_include,...

Error  acados_ocp  (line 181)
                ocp_compile_interface(obj.opts_struct);

Error minimal_example_ocp (line 107)
ocp = acados_ocp(ocp_model, ocp_opts);`
What should I do about this error?
I tried Acados on an Ubuntu virtual machine and it worked

FreyJo, 2021-09-22:
Can you check in the `mex` call in https://github.com/acados/acados/blob/master/interfaces/acados_matlab_octave/ocp_compile_interface.m
if the paths that are arguments for `mex` are correct?

Faust-Wang, 2021-09-22:
Thank you for your reply. How can I check it, please?  Can you give a routine? I need mex(??????).

FreyJo, 2021-09-22:
I guess you are using Matlab and the error occurs here:
https://github.com/acados/acados/blob/master/interfaces/acados_matlab_octave/ocp_compile_interface.m#L169
So, set a break point before that and check the arguments of this `mex` call.

Faust-Wang, 2021-09-22:
Yes. The initial error indicates that the associated library cannot be found.1.Error: MEX could not find the library 'acados' specified with the -l option. MEX has searched for files with one of the following names: libacados.a acados.lib libacados.lib Verify that the library name is correct. If the library is not there In an existing path, use the -l option to specify the path.
I modify line 169 of 'ocp_compile_interface.m' (<'-lacados', '-lhpipm', '-lblasfeo',libs.osqp> to <'-Lacados', '-Lhpipm', '-Lblasfeo','-Losqp'>).
Ok, this error is gone. But the second erroecame out and I couldn't fix it.
Now, The parameter is `mex(mex_flags,` FLAGS, LDFLAGS, COMPDEFINES, COMPFLAGS, acados_include, acados_interfaces_include, external_include, blasfeo_include, hpipm_include, acados_lib_path, '-Lacados', '-Lhpipm', '-Lblasfeo', libs.qpoases,libs.qpdunes, '-Losqp', libs.hpmpc, libs.ooqp, mex_files{ii}, '-outdir', opts.output_dir)`
However, a new error has emerged as mentioned above

Faust-Wang, 2021-09-22:
I checked all the parameters as follows:
`mex_flags`：An empty 0 by 0 char array =============I guess there's something wrong with  `mex_flags = getenv('ACADOS_MEX_FLAGS')` 
`FLAGS`: 'CFLAGS=$CFLAGS -std=c99 -DACADOS_WITH_OSQP'
`LDFLAGS`:  'LDFLAGS=$LDFLAGS'
`COMPDEFINES`: 'COMPDEFINES=$COMPDEFINES -DACADOS_WITH_OSQP'
`COMPFLAGS`: 'COMPFLAGS=$COMPFLAGS'
`acados_include`: '-IE:\Project\acados\examples\acados_matlab_octave\..\..'
`acados_interfaces_include`:  '-IE:\Project\acados\examples\acados_matlab_octave\..\..\interfaces'
`external_include`: '-IE:\Project\acados\examples\acados_matlab_octave\..\..\external'
``:
``:
``:

FreyJo, 2021-09-23:
Ah, I see, you should not do this:
> I modify line 169 of 'ocp_compile_interface.m' (<'-lacados', '-lhpipm', '-lblasfeo',libs.osqp> to <'-Lacados', '-Lhpipm', '-Lblasfeo','-Losqp'>).

from the gcc docu, how -L and -l should be used:
```
    gcc source.c --output executable -Iheader_path -Llibrary_path -llibrary_name
```

Did you check what is in your folder `Project\acados\lib`, i.e. if all libs are compiled successfully?

Faust-Wang, 2021-09-24:
As you said, the lib folder has only one file `link_libs.json`.
However, following the two steps of the tutorial(`https://docs.acados.org/installation/index.html`), no errors were reported:
`cmake.exe -G "MinGW Makefiles" -DACADOS_INSTALL_DIR="$ACADOS_INSTALL_DIR" -DBUILD_SHARED_LIBS=OFF -DACADOS_WITH_OSQP=ON ..`
`mingw32-make.exe -j4`
`mingw32-make.exe install`
These operations were all successful.
Which lib should I compile separately? Whether to use this command `gcc source.c --output executable -Iheader_path -Llibrary_path -llibrary_name`

FreyJo, 2021-09-24:
It seems like passing the argument "-DACADOS_INSTALL_DIR="$ACADOS_INSTALL_DIR"" to CMake didn't work correctly.
You can of course manually copy the lib files into the folder `acados/lib`.

Faust-Wang, 2021-09-24:
This is true. This example is ready to run.
But I still get this -L error when compiling other project files.
error: `MEX could not find the library 'cl_swarming' specified with the -l option.  
MEX has searched for files with one of the following names:  
libcl_swarming.a  
cl_swarming.lib  
libcl_swarming.lib  
Verify that the library name is correct.  If the library is not there  
In an existing path, use the -l option to specify the path. `
I checked and found that the C code matlab generated by Casadi could not compile into a.lib file.  So acados can't call.lib  
Do you have a solution for this situation

FreyJo, 2021-09-24:
> I checked and found that the C code matlab generated by Casadi could not compile into a.lib file.

Why is that? What was the error when compiling the model?

Faust-Wang, 2021-09-24:
> > I checked and found that the C code matlab generated by Casadi could not compile into a.lib file.
> 
> Why is that? What was the error when compiling the model?

The error is very similar to the one mentioned above：error: `MEX could not find the library 'cl_swarming' specified with the -l option. MEX has searched for files with one of the following names: libcl_swarming.a cl_swarming.lib libcl_swarming.lib Verify that the library name is correct. If the library is not there In an existing path, use the -l option to specify the path.` 
However, I can compile `cl_swarming.so` files automatically on Ubuntu.
Therefore, I would like to ask: how should I handle the error with ` the library is not there In an existing path, use the -l option to specify the path`?

FreyJo, 2021-09-28:
Is the model lib compiled successfully?
I.e. is a library with one of the names `libcl_swarming.a cl_swarming.lib libcl_swarming.lib` in your `build` folder?
If not, there must have been an error before the one you posted..

Faust-Wang, 2021-10-07:
Unfortunately, I still haven't compiled successfully.  I don't know why ` lib` doesn't compile.  On Ubuntu I had no trouble compiling the `so` .

closed 2021-12-09 by FreyJo