acados/acados #765: Make MEX wrapper around templated solver work on Windows

FreyJo, 2021-11-02:
Just to keep track of this, since it would be useful for all Simulink and Windows users.
current error: https://pastebin.com/hPCTV6KY
probably templated Makefile needs to be adapted

From https://github.com/acados/acados/pull/712

FreyJo, 2023-01-12:
TODO:
try switching to templated CMake and see if / how compatible it is with Windows matlab!

AkashJohnSubash, 2023-05-15:
@sourishpramanick You got this.

FreyJo, 2023-06-01:
Proof of concept PR https://github.com/acados/acados/pull/930 is merged.
Big thanks @yzuuang :fire_extinguisher: 
Let's remove the backend of the old Matlab interface completely and make it use the templates!
For reference on the duplicated backends we have now:
https://github.com/acados/acados/blob/2a0c1e20072a7014a74e337728ee13a01c3a2204/docs/matlab_octave_interface/mex_interface_overview.png
https://docs.acados.org/matlab_octave_interface/index.html#interface-structure

closed 2023-06-16 by FreyJo