acados/acados #770: C-code is not generated correctly in Matlab

MuhammadAlhaddad, 2021-12-05:
Dear all! I am trying to generate c_code from Matlab script for NMPC problem. The minimal_example_ocp.m works fine and generates c code which also works well. However, when I try to generate c code from my matlab nmpc script, the c code file is generated with this warning message:
```
Warning: cost_y_ref not defined for ocp json.
Using zeros(ny,1) by default. 
> In set_up_acados_ocp_nlp_json (line 412)
In acados_ocp/generate_c_code (line 227)
In my_mpc2 (line 157) 
Warning: cost_y_ref_0 not defined for ocp json.
Using zeros(ny_0,1) by default. 
> In set_up_acados_ocp_nlp_json (line 429)
In acados_ocp/generate_c_code (line 227)
In my_mpc2 (line 157) 
Warning: cost_y_ref_e not defined for ocp json.
Using zeros(ny_e,1) by default. 
> In set_up_acados_ocp_nlp_json (line 445)
In acados_ocp/generate_c_code (line 227)
In my_mpc2 (line 157) 
```
The results from the generated c code is widely different from the right results in Matlab. While the results of minimal_example_ocp are the same in Matlab and c-generated_code. As results I mean xtraj and utraj.

FreyJo, 2021-12-07:
I think the warnings are very descriptive.

> The results from the generated c code is widely different from the right results in Matlab. While the results of minimal_example_ocp are the same in Matlab and c-generated_code. As results I mean xtraj and utraj.

How do you run the generated C code?

On Linux, you can interact with the template based solver in Matlab, similar to the native Matlab solver.
This is tested for some problems, e.g. here:
https://github.com/acados/acados/blob/master/examples/acados_matlab_octave/test/test_template_lin_mass_ocp.m

If you can produce an issue similar to that test file, I can look into it.

MuhammadAlhaddad, 2021-12-14:
@FreyJo  Thanks for reply and sorry for late response. Actually I am new to acados. I run C code immediately from the terminal.
For example I have the follwoing results in Matlab

```
iter	res_stat	res_eq		res_ineq	res_comp	qp_stat	qp_iter
0	1.476362e+04	0.000000e+00	1.000000e+00	0.000000e+00	0	0
1	9.318043e+03	2.220446e-16	2.220446e-16	1.783073e-12	0	9
2	7.498445e+03	1.952317e-01	2.220446e-16	1.392360e-09	0	17
3	2.017743e+06	2.032719e-01	2.420730e-12	4.299985e-07	0	11
4	1.929637e+05	7.647044e-01	2.220446e-16	1.070766e-08	0	18
5	1.592372e+04	3.768312e-01	2.220446e-16	5.910387e-10	0	15
6	1.229219e+03	4.238779e-01	2.220446e-16	2.694797e-13	0	10
7	1.065885e+03	2.269782e-01	2.220446e-16	9.417510e-12	0	9
8	5.888070e+02	1.121430e-01	2.220446e-16	4.112027e-14	0	8
9	4.092439e+02	1.960283e-01	2.220446e-16	2.735053e-12	0	9
10	3.351594e+02	2.473606e-02	2.220446e-16	1.881060e-10	0	7
11	4.430100e+02	3.686669e-02	2.220446e-16	1.625746e-12	0	7
12	3.770445e+02	1.902244e-02	2.220446e-16	1.905883e-09	0	7
13	4.546806e+02	9.262698e-03	2.220446e-16	5.863920e-08	0	6
14	4.145853e+02	1.777634e-02	2.220446e-16	6.397591e-11	0	8
15	3.754686e+02	2.206567e-02	2.220446e-16	1.117293e-10	0	7
16	3.293133e+02	1.817794e-02	2.220446e-16	4.363151e-09	0	7
17	3.683309e+02	1.371988e-02	2.220446e-16	4.510556e-08	0	6
18	3.073112e+02	1.379703e-02	2.220446e-16	9.041309e-10	0	7
19	3.490917e+02	1.242016e-02	2.220446e-16	8.541869e-08	0	6
20	2.940414e+02	1.185934e-02	2.220446e-16	3.384381e-10	0	7
21	3.325341e+02	1.090496e-02	2.220446e-16	1.304420e-07	0	6
22	2.827022e+02	1.052844e-02	2.220446e-16	1.214574e-10	0	7
23	3.184843e+02	9.979626e-03	2.220446e-16	9.165332e-08	0	6
24	2.729342e+02	9.624133e-03	2.220446e-16	4.806401e-11	0	7
25	3.064890e+02	9.240959e-03	2.220446e-16	6.190015e-08	0	6
26	2.644605e+02	8.925942e-03	2.220446e-16	2.139454e-11	0	7
27	2.961477e+02	8.646951e-03	2.220446e-16	4.326006e-08	0	6
28	2.570477e+02	8.361607e-03	2.220446e-16	1.026300e-11	0	7
29	2.871440e+02	8.149950e-03	2.220446e-16	3.124764e-08	0	6
30	2.505095e+02	7.889948e-03	2.220446e-16	4.773287e-12	0	7
31	2.792338e+02	7.726518e-03	2.220446e-16	2.316427e-08	0	6
32	2.446997e+02	7.487960e-03	2.220446e-16	2.520729e-12	0	7
33	2.722284e+02	7.360755e-03	2.220446e-16	1.767444e-08	0	6
34	2.395026e+02	7.140612e-03	2.220446e-16	1.402764e-12	0	7
35	2.659808e+02	7.041546e-03	2.220446e-16	1.383014e-08	0	6
36	2.348264e+02	6.837339e-03	2.220446e-16	8.181396e-13	0	7
37	2.603747e+02	6.760621e-03	2.220446e-16	1.106258e-08	0	6
38	2.305972e+02	6.570304e-03	2.220446e-16	4.976605e-13	0	7
39	2.553169e+02	6.511628e-03	2.220446e-16	9.020680e-09	0	6
40	2.267547e+02	6.333489e-03	2.220446e-16	3.143635e-13	0	7
41	2.507319e+02	6.289567e-03	2.220446e-16	7.481118e-09	0	6
42	2.232494e+02	6.122165e-03	2.220446e-16	2.054406e-13	0	7
43	2.465579e+02	6.090442e-03	2.220446e-16	6.297798e-09	0	6
44	2.200399e+02	5.932555e-03	2.220446e-16	1.384400e-13	0	7
45	2.427434e+02	5.911013e-03	2.220446e-16	5.372622e-09	0	6
46	2.170917e+02	5.761595e-03	2.220446e-16	9.591835e-14	0	7
47	2.392457e+02	5.748622e-03	4.440892e-16	4.641803e-09	0	6
48	2.143755e+02	5.606778e-03	2.220446e-16	6.815593e-14	0	7
49	2.360284e+02	5.601070e-03	8.881784e-16	4.053857e-09	0	6
50	0.000000e+00	0.000000e+00	0.000000e+00	0.000000e+00	0	7
```
I run the c code as in pictures follow and I have the shown results
[image]

[image]

Thanks in advance

FreyJo, 2021-12-15:
It is still not clear to me how you use the C code.
As in https://github.com/acados/acados/blob/master/examples/acados_matlab_octave/test/test_template_lin_mass_ocp.m
Or do you run the generated main file?
The latter is just to show how the solver can be used.
It is not expected to give the same result as the solver in Matlab, since the input data still has to be set.

MuhammadAlhaddad, 2021-12-18:
Hi @FreyJo !
Actually I was trying to run the executable file which is generated without any modification. However, now I am working on calling the solver in Matlab as i have understood from your responces. I was able to run https://github.com/acados/acados/blob/master/examples/acados_matlab_octave/test/test_template_lin_mass_ocp.m and all other test files in acados_matlab_octave/test folder. For script mine, I have generated c code succesfully using `ocp.generate_c_code`. But when i tired to call the solver using `t_ocp = differential_drive_mex_solver` as in `test_template_lin_mass_ocp.m`, it gives me that there is no `libacados_ocp_solver_differential_drive.so` even though it has been generated in c_generated_code folder as shown in screenshot. I am using Ubuntu 18.04. 
[image]

FreyJo, 2021-12-20:
I think you need to make sure that the `c_generated_code` folder is also exported.
In the test files I mentioned, this is done here: https://github.com/acados/acados/blob/91a01d4c8db13b5de0d7c600eec72a96b3b5d771/examples/acados_matlab_octave/test/env.sh#L76
Please try to use the `env.sh` file from `examples/acados_matlab_octave/test`.

MuhammadAlhaddad, 2021-12-20:
@FreyJo Thank you! This fix the problem mine. I didn't export it before. Could I ask you how I can set diffrenent values for `y_ref` when I test the solver?
For example when I creat solver I put the reference points for every node as following
```
for k=0:N-1
   yref = path(k+1,:)';
   ocp.set('cost_y_ref', yref, k);
end
```
But if I want to modify this reference states when I test the created solver, I couldn't put one `y_ref` value for every `k`. The solver accept only
`t_ocp.set('cost_y_ref' , y_ref)`.

In addition, there is something that I couldn't understand it. The total time `time_tot` is about 10 times more when I call the solver than when I created the solver as it is shown in picture.
[image]

MuhammadAlhaddad, 2021-12-20:
> Why does setting y_ref stage wise not work for you?
My fault was that is the for loop up to model.N not model.N-1

> There should be no real difference in the CPU times.
> Are those the same solver calls, i.e. with the same initialization?

Actually I didn't add any initialization when I call the solver so the time total was longer. Thanks alot Jonathan!

FreyJo, 2021-12-20:
Great, I am closing this one then!

closed 2021-12-20 by FreyJo

MuhammadAlhaddad, 2021-12-23:
@FreyJo  Hi Jonathan! I have a question please. Can I change the `cost_expr_y` or `cost_expr_y_e` when I call the created solver? This for the cost function type `nonlinear_ls`. My goal is to add some new terms to the cost function when I call it.  I have seen that t_ocp only support the following values.  I have tried to add these terms using cost_Vx but it doesn't support casadi formulas.
```
Error using acados_mex_set_differential_drive
supported values are:
p, constr_x0, constr_lbx, constr_ubx, constr_C, constr_D, constr_lg, constr_ug, constr_lh, constr_uh constr_lbu, constr_ubu,
cost_y_ref[_e], cost_Vu, cost_Vx, cost_Vz, cost_W, cost_Z, cost_Zl, cost_Zu, cost_z, cost_zl, cost_zu, init_x, init_u, init_z,
init_xdot, init_gnsf_phi, init_pi, nlp_solver_max_iter, qp_warm_start, warm_start_first_qp, print_level
```