acados/acados #777: [Errno 9] in Python when installing acados with qpDUNES

FreyJo, 2021-12-22:
I had weird errors using the test files added in PR #772

I install the acados shared library with CMake on Ubuntu 20.04.
When I install acados without qpDUNES the newly added example / test files, work as expected.
When installing with qpDUNES, I get the following error.
```
  File "marathos_test_problem.py", line 243, in <module>
    main()
  File "marathos_test_problem.py", line 59, in main
    solve_marathos_problem_with_setting(setting)
  File "marathos_test_problem.py", line 171, in solve_marathos_problem_with_setting
    ocp_solver.print_statistics()
  File "/home/oj/acados/interfaces/acados_template/acados_template/acados_ocp_solver.py", line 1052, in print_statistics
    print('\niter\tres_stat\tres_eq\t\tres_ineq\tres_comp\tqp_stat\tqp_iter\talpha')
OSError: [Errno 9] Bad file descriptor
Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
OSError: [Errno 9] Bad file descriptor
```
Somehow stdout is broken, when executing the second test run.
Basically, any Python `print` after exiting the function in which `AcadosOcpSolver` lives, fails on my machine.

The same error also occurred on Github Actions, see https://github.com/acados/acados/runs/4605115813?check_suite_focus=true
```
Traceback (most recent call last):
  File "marathos_test_problem.py", line 243, in <module>
    main()
  File "marathos_test_problem.py", line 59, in main
    solve_marathos_problem_with_setting(setting)
  File "marathos_test_problem.py", line 171, in solve_marathos_problem_with_setting
    ocp_solver.print_statistics()
  File "/home/runner/work/acados/acados/acadosenv/lib/python3.8/site-packages/acados_template/acados_ocp_solver.py", line 1056, in print_statistics
    print(f'{int(stat[0][jj]):d}\t{stat[1][jj]:e}\t{stat[2][jj]:e}\t{stat[3][jj]:e}\t' +
OSError: [Errno 9] Bad file descriptor
```

Both locally and on Github Actions, this is fixed when compiling `acados` without qpDUNES.
https://github.com/acados/acados/pull/772/commits/cce9d8bbb0e47f5e942c2ea98c990b367b3c9a9e

Any ideas what could be the issue are appreciated!

giaf, 2021-12-26:
I'm not too deep in python so not really idea why this happens.
But just searching on the web, there may be some hint of what kind of issues may cause this, and give some ideas on where to look for possible issues on the python or C qpDUNES wrappers.

schwieni, 2022-02-13:
Explanation: the default behavior of qpDunes is to close stdout during destruction. See:
- `external/qpdunes/include/MessageHandling.cpp:301` in destructor `~MessageHandling()` : `fclose( outputFile );`
- `external/qpdunes/include/MessageHandling.cpp:229` in constructor `MessageHandling()` : `outputFile = stdFile;`
- `external/qpdunes/include/qpdunes/MessageHandling.hpp` definition `static FILE* stdFile = stdout;`

This might be fixed in qpDunes by adding the check `outputFile != stdout` in the destructor of class `MessageHandling` before closing the file.

PR #785 should fix the issue in acados directly.
The solution is to load `libacados.so` separately to avoid destruction.

FreyJo, 2022-02-15:
Thanks again for the PR and the detailed explanation!

closed 2022-02-15 by FreyJo