acados/acados #787: Exception when using globalization

MuhammadAlhaddad, 2022-02-19:
Hi! I tried to use globalization in solver option but it didn't work so i tried some codes which also contain it in` example/acados_python` and it gives me the same issue.
I have tried this code
`~/acados/examples/acados_python/crane$ python3 time_optimal_example.py`
and it gives me the exception
```
Traceback (most recent call last):
  File "/home/muhammad/MPC_CollisionAvoidance/catkin_ws/src/nmpc_ca/acados/interfaces/acados_template/acados_template/utils.py", line 282, in ocp_check_against_layout_recursion
    layout_of_key = layout[key]
KeyError: 'globalization'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "time_optimal_example.py", line 120, in <module>
    ocp_solver = AcadosOcpSolver(ocp, json_file = 'acados_ocp.json')
  File "/home/muhammad/MPC_CollisionAvoidance/catkin_ws/src/nmpc_ca/acados/interfaces/acados_template/acados_template/acados_ocp_solver.py", line 679, in __init__
    ocp_formulation_json_dump(acados_ocp, json_file)
  File "/home/muhammad/MPC_CollisionAvoidance/catkin_ws/src/nmpc_ca/acados/interfaces/acados_template/acados_template/acados_ocp_solver.py", line 441, in ocp_formulation_json_dump
    ocp_check_against_layout(ocp_nlp_dict, dims_dict)
  File "/home/muhammad/MPC_CollisionAvoidance/catkin_ws/src/nmpc_ca/acados/interfaces/acados_template/acados_template/utils.py", line 272, in ocp_check_against_layout
    ocp_check_against_layout_recursion(ocp_nlp, ocp_dims, ocp_nlp_layout)
  File "/home/muhammad/MPC_CollisionAvoidance/catkin_ws/src/nmpc_ca/acados/interfaces/acados_template/acados_template/utils.py", line 288, in ocp_check_against_layout_recursion
    ocp_check_against_layout_recursion(item, ocp_dims, layout_of_key)
  File "/home/muhammad/MPC_CollisionAvoidance/catkin_ws/src/nmpc_ca/acados/interfaces/acados_template/acados_template/utils.py", line 285, in ocp_check_against_layout_recursion
    " '{0}' is not in layout but in OCP description.".format(key))
Exception: ocp_check_against_layout_recursion: field 'globalization' is not in layout but in OCP description.
```
In my code when i remove globalization option, the solver works fine.

FreyJo, 2022-02-19:
The example is tested on Github Actions.
Did you do a fresh install of the latest version?

MuhammadAlhaddad, 2022-02-19:
@FreyJo  Thanks Jonathan! i had an old version from another repos.

closed 2022-02-19 by MuhammadAlhaddad