acados/acados #788: EXTERNAL cost + GAUSS_NEWTON Hessian

zanellia, 2022-02-22:
Am I missing something or one could in principle set up a solver and an OCP with 
`ocp.solver_options.hessian_approx = 'GAUSS_NEWTON'`
and
`ocp.cost.cost_type = 'EXTERNAL'`
?
Would it not make sense to add a check that rules out that these two options are not used together? After a quick look, I think the C code is actually ignoring the GAUSS_NEWTON option, but still, it is rather misleading.

FreyJo, 2022-03-14:
I vaguely remember a discussion about this.
I think the conclusion was that for `EXTERNAL` cost module GAUSS_NEWTON hessian is not defined and thus computing the hessian for it exactly is the only option.
Maybe it makes sense to document this here https://github.com/acados/acados/blob/master/interfaces/acados_template/acados_template/acados_ocp.py#L2168
Or print some warning here:
https://github.com/acados/acados/blob/master/interfaces/acados_template/acados_template/acados_ocp_solver.py#L60

zanellia, 2022-03-15:
> I think the conclusion was that for EXTERNAL cost module GAUSS_NEWTON hessian is not defined and thus computing the hessian for it exactly is the only option.

Absolutely. The thing is that allowing for `GAUSS_NEWTON` some users might be tricked into thinking 1) they are actually using Gauss-Newton (not true and impossible) or 2) this is a bug and leads to undefined behavior (I am saying this because I encountered this situation with a couple of users already).

FreyJo, 2022-03-15:
Agreed, I added a warning now!

closed 2022-03-15 by FreyJo