acados/acados #798: Installation for Windows

BuwuliaoDEhenwuliao, 2022-03-19:
I was installing Acados for Windows, but when I run`mingw32-make.exe -j4`,  error occurs:
`[ 84%] Building C object acados/CMakeFiles/acados.dir/utils/timing.c.obj                                                D:\DATA\Robot\acados\acados\utils\timing.c: In function 'acados_tic':                                                   D:\DATA\Robot\acados\acados\utils\timing.c:117:36: warning: implicit declaration of function 'clock_gettime' [-Wimplicit-function-declaration]                                                                                                   void acados_tic(acados_timer* t) { clock_gettime(CLOCK_MONOTONIC, &t->tic); }                                                                              ^~~~~~~~~~~~~                                                                       D:\DATA\Robot\acados\acados\utils\timing.c:117:50: error: 'CLOCK_MONOTONIC' undeclared (first use in this function)      void acados_tic(acados_timer* t) { clock_gettime(CLOCK_MONOTONIC, &t->tic); }                                                                                            ^~~~~~~~~~~~~~~                                                       D:\DATA\Robot\acados\acados\utils\timing.c:117:50: note: each undeclared identifier is reported only once for each function it appears in                                                                                                       D:\DATA\Robot\acados\acados\utils\timing.c: In function 'acados_toc':                                                   D:\DATA\Robot\acados\acados\utils\timing.c:123:19: error: 'CLOCK_MONOTONIC' undeclared (first use in this function)          clock_gettime(CLOCK_MONOTONIC, &t->toc);                                                                                              ^~~~~~~~~~~~~~~                                                                                      acados\CMakeFiles\acados.dir\build.make:645: recipe for target 'acados/CMakeFiles/acados.dir/utils/timing.c.obj' failed mingw32-make[2]: *** [acados/CMakeFiles/acados.dir/utils/timing.c.obj] Error 1                                          CMakeFiles\Makefile2:360: recipe for target 'acados/CMakeFiles/acados.dir/all' failed                                   mingw32-make[1]: *** [acados/CMakeFiles/acados.dir/all] Error 2                                                         Makefile:144: recipe for target 'all' failed                                                                            mingw32-make: *** [all] Error 2  `

What should I do about this error?
Thank you in advance.

FreyJo, 2022-03-21:
Hi,

I was looking around a bit what could be the source, but I am not really sure.
I am not very familiar with the timing code for the different operating systems, but on recent Windows Matlab installations this is not an issue.

One thing that should definitely work is adding the flag `-DCMAKE_BUILD_TYPE=Debug` to the CMake command. This turns off the acados timings.

For a proper fix, these links could be helpful:
- https://stackoverflow.com/questions/29666937/error-clock-monotonic-undeclared-first-use-in-this-function
- https://github.com/acados/acados/commit/90d75386cad9f2b16115cce04685e90934c0f7d8 maybe something is wrong in this commit?
- It would make sense to double check the logic in this file https://github.com/acados/acados/blob/master/acados/utils/timing.c but it seems that the expected branch of the file is being compiled.

ZionLIU-hit, 2022-03-21:
Hi Jonathan,
Thank you for your advice and I have completed the installation with `-DCMAKE_BUILD_TYPE=Debug`.
But when I try the example "minimal_example_ocp", I get the messages:
`C:/ProgramData/MATLAB/SupportPackages/R2021a/3P.instrset/mingw_w64.instrset/bin/../lib/gcc/x86_64-w64-mingw32/6.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe:
skipping incompatible D:\DATA\Robot\acados\examples\acados_matlab_octave\..\..\lib\acados.lib when searching for -lacados
C:/ProgramData/MATLAB/SupportPackages/R2021a/3P.instrset/mingw_w64.instrset/bin/../lib/gcc/x86_64-w64-mingw32/6.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe:
cannot find -lacados
collect2.exe: error: ld returned 1 exit status`
and the same messages for blasefeo, hpipm, qpOASES_e, qpdunes.
When I use the dumpbin.exe to examine the acados.lib, I find that it is 32-bit, so I use MINGW-i686-8.1.0-release-win32-sjlj, but the erroe remains.

Are there anybody could help me?
Thanks again.

closed 2022-04-25 by ZionLIU-hit