acados/acados #811: Declaration of XXX Changes Meaning of ‘ocp_qp_solver_plan’ with -Wpedantic

ErcBunny, 2022-04-17:
Greetings~

I get the following errors with the `-Wpedantic` option (with ROS2 Colcon building tool). A quick fix is to switch off the "pedantic" option.

```
error: declaration of ‘ocp_qp_solver_plan ocp_nlp_plan::ocp_qp_solver_plan’ changes meaning of ‘ocp_qp_solver_plan’ [-fpermissive]
note: ‘ocp_qp_solver_plan’ declared here as ‘typedef struct ocp_qp_solver_plan ocp_qp_solver_plan’

error: declaration of ‘sim_solver_plan* ocp_nlp_plan::sim_solver_plan’ changes meaning of ‘sim_solver_plan’ [-fpermissive]
note: ‘sim_solver_plan’ declared here as ‘typedef struct sim_solver_plan sim_solver_plan’
```

I wonder if this is a known bug. Is it OK to proceed if I just suppress these warnings?

FreyJo, 2022-04-19:
Are you on the latest acados version?
I think this has been addressed here: https://github.com/acados/acados/pull/790

ErcBunny, 2022-04-22:
I am currently on 0.1.7. Thanks for linking me to the PR. I think I'll wait for the next pre-release version and use the work-around for now, since the solver works just fine.

FreyJo, 2022-07-21:
By now there is a new release which contains the fix.