acados/acados #822: swarming   solutionfailed

johnwaneboy, 2022-05-19:
i run the demo of   example_sim.m, example_ocp.m, example_closed_loop.m      in acados/examples/acados_matlab_octave/swarming
 but it always solution  failed ~~~~

johnwaneboy, 2022-05-19:
[image]

FreyJo, 2022-05-19:
On which acados version are you?
What error do you get when running `example_sim`?

johnwaneboy, 2022-05-19:
example_sim  could run sucessfully
[image]

FreyJo, 2022-05-19:
ok, just the `example_ocp` doesn't work for you.
Can you switch to the latest acados version and check if it works?
The version you have seems at least 6 months old.

johnwaneboy, 2022-05-19:
thanks ！ I will try later, and I have build the the same model in python + ubuntu, but if I set the path constraints（con_h_expr） such as the minimum distance between agents, then solution will fail, even I set the lh to infinitesimal and the uh to infinity.

johnwaneboy, 2022-05-19:
I have switched to the latest version， the example_closed_ loop.  and   example_ ocp.m stil solution failed.

FreyJo, 2022-05-19:
Did you also update the submodules?
What OS are you using?
Can you copy the output of the failing example here?

johnwaneboy, 2022-05-19:
Ihave reinstalled the acados,  my environment is ubuntu 18.06 with python 3.6. 
when i run my demo, if there is no constraints, it works well, but when i add constraint, it return status 4

[image]

johnwaneboy, 2022-05-20:
Thank you very much for helping solve the problem . What other information can I supply to help you solve the problem ?

sandmaennchen, 2024-05-06:
Is this still an issue? The example works for me using the current acados version.

closed 2024-05-06 by sandmaennchen